magicbug/Cloudlog #1455: LOTW sync issues: not detecting previous QSOs // can't download from LOTW

tgreaves, 2022-04-10:
Hi,

I'm having issues with syncing with LOTW.

I have imported my certificate successfully.  However, when I go to sync, I get the following report:

```
M7TGZ (Home QTH) No QSOs to Upload

LoTW Matches
LoTW Downloading failed either due to it being down or incorrect logins.
```

However. I have 120+ QSOs which are existing which have not been uploaded. (And who as 'Sent to LOTW as No).

How can I debug why these QSOs are not being picked up?  They predate when I added the Certificate BUT the Certificate QSO start date is appropriately in the past.

Regarding the second error, I have confirmed that the username/password is correct and allows me to login to the LOTW website.

As a frame of reference, QRZ syncing is working perfectly.

Thanks for your help!

magicbug, 2022-04-10:
Hi,

As you have said the p12 certs have been uploaded would be useful to see some screenshots of both the LoTW area and the station location as the data needs to match.

As for the downloading matches unless CURL is missing on the system there's not a huge amount to go wrong, you can enable full debugging in /applcation/config/config.php and run sync again and check application/logs to see if it contains any hints but usually this failing is purely down to login data being wrong.

tgreaves, 2022-04-10:
As an update, I:

* Manually exported (ADIF) all my log entries, then signed/uploaded them using TQSL.
* Syncing with CloudLog then started to work (I didn't adjust my credentials).  I.e. CloudLog would download matches from LOTW successfully.

So it looks like CloudLog has challenges when trying to upload to the first time to an empty LOTW logbook.  Once TQSL is used to populate the logbook, CloudLog springs into life...

magicbug, 2022-04-10:
Not quite sure I fully understand, you'll only get Matches when there's QSOs uploaded to LoTWs award system, if that's working now that's great however you first said the issue was "uploading qsos" from Cloudlog?

tgreaves, 2022-04-10:
In short: Cloudlog fails to upload QSOs directly to LoTW if the LoTW logbook on their systems is empty.  The process fails with the error messages in my original post.

Once the LoTW logbook on their system has entries (That I uploaded _manually_ by exporting from Cloudlog and then using TQSL), Cloudlog starts to work with syncing.

So there seems to be a bug preventing Cloudlog from uploading or syncing with LoTW when the LoTW logbook is empty.

Does that help?

magicbug, 2022-04-10:
Does but cant see it being a fault my side, when Cloudlog uploads to LoTW it just gets a failed or / success which your first message didn't report just `M7TGZ (Home QTH) No QSOs to Upload` this suggests Cloudlog didn't find anything that matched the certificate.

`LoTW Downloading failed either due to it being down or incorrect logins.`

this could be possibly due to no QSOs being in LoTW when a downloads requested, but first time I've seen it reported in two years.

magicbug, 2022-04-10:
I'm going to take it that both functions are now working? should report QSOS uploaded and matches found

tgreaves, 2022-04-10:
Thanks for coming back!

It will upload _new_ QSOs but refused to upload ones that existed prior to me uploading the certificate.  So I had to manually upload those by downloading them and submitting with TQSL.

It would be good if there was a way to also upload previous QSOs in this situation, but I found a way round it.

closed 2022-04-10 by magicbug