magicbug/Cloudlog #1464: Distance in QSO entry screen calculated from user account locator, not station locator locator

mpentler, 2022-04-24:
**Describe the bug**
When entering a locator square for a QSO the distance to you is calculated from your username’s locator square entry, rather than the locator square entry of the station location. This didn’t seem right.

**To Reproduce**
1) enter your locator square incorrectly in your user account settings when setting up cloudlog (as I did to accidentally discover this behaviour)
2) enter a QSO, in my case a Post QSO entry
3) the distance is calculated to your incorrect locator square

**Expected behaviour**
The distance is calculated to the station location’s locator square, as you’d want when operating /P - is it uses the user name’s locator entry then this seems to not be optimal behaviour.

**Smartphone:**
 - Device: iPhone 11
 - OS: 15.4.1
 - Browser Safari
 - latest

magicbug, 2022-04-24:
I've a feeling this has been resolved in the dev branch currently named station_logbooks but might need to double check

mpentler, 2022-04-24:
Yes I did just find #1408

closed 2022-04-24 by magicbug