magicbug/Cloudlog #1465: I5 call misplaced on map

pa8s, 2022-04-25:
**Describe the bug**
When logging a qso with I5 prefix this qso is marked on the map as being located in Mogadishu/Muqdisho Somalia.

**To Reproduce**
1. Create log entry with an I5 prefix
2. Look at the overview map near Somalia, there is the mark.
3. Double click on the call in the logbook and QSO data window pops up.
4. Here the mark on the map is in Rome. 

**Expected behaviour**
Mark on the overview map somewhere in Tuscany or just Italy.

**Screenshots**
Overview Map
[image]

QSOData Map
[image]

**Server**
 - OS: FreeBSD 12
 - HTTP Server: Apache 2.Firefox 99
 - Version: 073

**Desktop**
 - OS: Linux Ubuntu 20.04
 - Browser: Firefox 99
 - Version: 073

**Additional context**
Assuming a typo in the code as Somalia is prefix T5

magicbug, 2022-04-25:
Just tested this here on the master branch and it places the callsign in Italy so I'm unable to see what the issue might be.

Have you I5TEST QSOs in the log? does the QSO contain any gridsquares.. I'd check the edit area for any rogue data

magicbug, 2022-04-25:
Poked around think i found the problem

phl0, 2022-04-25:
I can confirm that behaviour. Just not time to debug atm.

pa8s, 2022-04-25:
> Have you I5TEST QSOs in the log? does the QSO contain any gridsquares.. I'd check the edit area for any rogue data
No gridsquares in this log entry. Just the call.

magicbug, 2022-04-25:
Fixed

closed 2022-04-25 by magicbug

pa8s, 2022-04-25:
Just pulled the last revision. Issue still visible when looking at the [Custom Map](https://pa100thales.pa8s.nl/index.php/map/custom) and select start/end date 21-04-2022 / 25-04-2022 .

magicbug, 2022-04-25:
pull again and you'll find its also fixed

pa8s, 2022-04-25:
Super. Thank you for fixing it very quickly.

commit 4c2d21b5 (2022-04-25): Map fixes QSOs with only DXCC positions on map

Fixes #1465