magicbug/Cloudlog #1469: Off by 1 error in LoTW certificate Date Created

Vic152, 2022-05-03:
When the certificate's for LoTW `Date Created` is 02/05/2022 the QSOs for 02/05/2022 will not sync to the LoTW.
When the certificate's for LoTW `Date Created` is 02/05/2022 the QSOs for 03/05/2022 will  sync to the LoTW.

Observed: QSOs for 02/05/2022 were logged but after LoTW manual sync the following message was displayed: `No QSOs to Upload`
Expected: QSOs for 02/05/2022 were synched to LoTW

magicbug, 2022-05-03:
Sounds like a bug will get it sorted

Vic152, 2022-05-03:
It seems like `Date Created` corresponds in some unknown way with the LoTW cert `Valid` vs `QSO Dates`

magicbug, 2022-05-03:
Yeah its this https://github.com/magicbug/Cloudlog/blob/ff7523a300793c8188d76a11a9e4a29eb6c911c7/application/models/Logbook_model.php#L2568 that returns the QSOs will look into it in the morning

magicbug, 2022-05-03:
Oh just spotted, OK so when you create a lotw cert you give it a state you started which according to your cert is "2022-05-02 00:00:00" there for it will only upload from that Date as that's all LoTW will accept

Vic152, 2022-05-03:
It should accept all QSO from 01/07/2021

QSO Dates: 2021-07-01 00:00:00 - 2025-05-01 23:59:59

Vic152, 2022-05-03:
Using TQSL I was able to upload all those QSOs

[image]

magicbug, 2022-05-03:
OK shall check into this a bit more, I'm surprised ARRL issued a cert that begins later than the QSO date but its an odd system at best. at the moment Cloudlog uses the Begin date in the certficiate.

Vic152, 2022-05-03:
What I am saying is cert is valid from 02/05/2022 but I can upload QSOs from 01/07/2022.
I could not use that cert on the 01/05/2022 if I could travel in time.

Vic152, 2022-05-03:
> Oh just spotted, OK so when you create a lotw cert you give it a state you started which according to your cert is "2022-05-02 00:00:00" there for it will only upload from that Date as that's all LoTW will accept

Even if that were true it should still accept QSOs from 02/05/2022

Vic152, 2022-05-03:
After sync (or multiple synchs) the page is not updated. It's as if Cloudlog never acknowledged existence of those QSOs.

[image]

magicbug, 2022-05-03:
I believe it's true and as the codes available everyone can check for themselves.

However as stated will look tomorrow.

Vic152, 2022-06-10:
Hello, Any progress on this one?

hugosilvaparagramanet, 2022-08-24:
Was about the open an issue with this subject, I confirmed this behavior today. I have a certificate issued in July 2022 for QSOs from May 2022, and cloudlog would not sync the QSOs. 
I manually updated the `date_created` field in the table and it worked.
I can take a look at this in a few days. I guess what needs to be done is:
- create fields in the `lotw_certs` table for QSO Start Date and End Date (preload existing certificate data with `date_created` and `date_expires`)
- load the QSO Start Date and End Date when loading the certificate
- look at the new fields when selecting the QSOs to sync considering that  a) the certificate should be valid b) the QSO must be between `qso_start_date` and `qso_end_date`

anything else that I should pay attention to @magicbug  ?

phl0, 2022-08-24:
I may have a patch ready to address this issue. 

>     * create fields in the `lotw_certs` table for QSO Start Date and End Date (preload existing certificate data with `date_created` and `date_expires`)

Is done with a migration script which inserts the two columns, extracts the QSO start and end date from exisiting certificates and stores them in the new columns.

>     * load the QSO Start Date and End Date when loading the certificate

These values are read an imported into the new columns on uploading of a new LotW certificate.

>     * look at the new fields when selecting the QSOs to sync considering that  a) the certificate should be valid b) the QSO must be between `qso_start_date` and `qso_end_date`

First we check the (old) expiry date of the certificate itself. If not yet valid or expired we skip the signing and uploading process for that certificate. If the certificate is valid the QSO dates are checked against the QSO start and end date contained in the extensions of the certificate. If those are also valid the QSO is signed and uploaded to LotW.

The code is here:

https://github.com/magicbug/Cloudlog/compare/station_logbooks...phl0:Cloudlog:addQsoDatesToLotwCerts?expand=1

Checks, tests and feedback are more than welcome @hugosilvaparagramanet @Vic152. No warranty from my side though. HAve a backup at hand ;-)

@magicbug This PR will clash in combination with my running PR https://github.com/magicbug/Cloudlog/pull/1569. So if 1569 gets merged before this one I will update the number of the migration script here.

magicbug, 2022-08-24:
@phl0 I should have checked this reply first as I just merged it! I'll let you do another PR for this fix sorry!

But looks like this will resolve it :)

phl0, 2022-08-24:
> @phl0 I should have checked this reply first as I just merged it! I'll let you do another PR for this fix sorry!
> 
> But looks like this will resolve it :)

No worries. I will update my branch. Not PRred yet. Waiting for feedback.

phl0, 2022-08-24:
Branch updated from my side. Would like to see feedback by @hugosilvaparagramanet and/or @Vic152 before I make it a PR.

hugosilvaparagramanet, 2022-08-25:
I have tested @phl0 branch and everything worked correctly, the certificate was imported with the correct dates and later on Cloudlog catched "old" QSOs and uploaded them.
Thanks @phl0 !

magicbug, 2022-08-27:
Now resolved in the test branch :)

closed 2022-08-27 by magicbug