magicbug/Cloudlog #1473: No pins showing on custom map

pa8s, 2022-05-06:
**Describe the bug**
When selecting a time frame with logged QSO's (e.g. one week) and band to whatever suitable, no pins are showing on the custom map.

**To Reproduce**
1. Create Custom Map with valid time frame.
2. Select whatever band suitable (including 'Every band')
3. 'Load Map'
4. Bottom of screen shows: 'Showing QSOs for Custom Date for Active Station Profile - _name of profile_'
5. No pins visible on map (complete world map visible)

**Expected behaviour**
Pins for applicable QSO's visible on map.

**Screenshots**
N/A

**Server**

    OS: FreeBSD 12
    HTTP Server: Apache 2.4.53
    Cloudlog version: 073
    Git Pull performed on 2022/05/06 19:45Z

**Desktop**

    OS: Linux Ubuntu 20.04
    Browser: Firefox 99

**Additional context**
Checked this on two separate Cloudlog instances (for own and event call)

magicbug, 2022-05-07:
I suspect its because your selecting too many QSOs and servers timing out I just tried on a few copies here without much issue, Leaflet can handle about 2000 points becoming too slow and buggy.

closed 2022-06-13 by magicbug

Meganitrospeed, 2022-07-01:
Hello,

Im suffering from the same issue, with about 48 QSO´s.

Chrome dev log shows an issue with the changeState() function in leaflet

index.php:1 Uncaught SyntaxError: Unexpected token '<'
    at XMLHttpRequest.stateChanged (leafembed.js:64:58)

Running currently
	'char_set' => 'utf8mb4',
	'dbcollat' => 'utf8mb4_general_ci',

Im guesing its related to something in the database being in a wrong format. I have taken a look at it but not yet found a culprit.

Meganitrospeed, 2022-07-01:
This does however, not only affect custom maps. also the normal map* I will raise a new issue since the scope has changed