magicbug/Cloudlog #1483: Edit QSO shows ALL modes in mode dropdown list, not filtered

mpentler, 2022-05-20:
Not sure if by design or a bug, but if you do a Post or Live QSO log you only get your pre-activated mode list (pre-set up in the Modes section in Admin menu). If you then Edit a QSO the modes dropdown has absolutely every mode listed in there. I've had a few people confirm this behaviour on their setups.

I assume the intended behaviour is to show only the activated modes, but I wasn't sure.

AndreasK79, 2022-05-24:
I would say this is by design. If you have a logged QSO with a disabled mode (for some reason), the correct mode would not be displayed in edit mode (or saved), if the mode didn't exist in the dropdown list.

AndreasK79, 2022-05-24:
Yes, disabled post-log.