magicbug/Cloudlog #1485: Make the locator field always upper case like callsign

Cadair, 2022-05-25:
I would like to make the locator field always uppercase like the callsign field, is this something which would be a good PR?

If anyone knows where the bit of upcase magic for the callsign field is that would be handy :wink:

phl0, 2022-05-27:
Well I just tested on master and station_logbooks and it makes the chars be upper case always... ?!

magicbug, 2022-05-27:
It would be the model for adding a QSO that could have strtoupper added to the gridsquare field if its not there already

phl0, 2022-05-27:
Seems like there is some JS in place already that makes the input chars upper case on input.

magicbug, 2022-05-27:
I wonder if the issue is actually in the edit box Flo?

phl0, 2022-05-27:
Oh right. I was talking about the Live QSO input window. Shall check the others incl. the QSO edit ones.

magicbug, 2022-05-27:
I did too then started questioning my thinking :)

mpentler, 2022-05-27:
Some websites do display the second pair of letters in lowercase to disguish them somewhat, it is worth saying. Do we really want to do it? Is it convention for those last two digits to be displayed in that way?

magicbug, 2022-05-27:
I'm not aware of any fixed convention, personally, I always keep them all uppercase but that's a purely personal choice.

phl0, 2022-05-27:
> I wonder if the issue is actually in the edit box Flo?

Did a few more tests (all on station_logbooks): The QSO log windows (live and post) do convert to upper case on input instantly. Guess its some JS here. The edit QSO windows doesn't do anything like that and would also take lower case. owever the PHP code as some toUpper in here as call and grid are stored upper case only even if edited in lower case.

phl0, 2022-05-27:
Found it in the CSS code and fixed various minor issues. See commit https://github.com/magicbug/Cloudlog/commit/673dc0f6fbb5bafd97bccd4ee16304082a35c0c1. You might want to test that and report before I make it a PR.
On the log QSO section it uppercases fields callsign, locator, iota-ref, sota-ref and darc_dok. On the edit QSO section it also uppercases vucc_grids besides the aforementioned. Happy to hear from your results/tests.

Cadair, 2022-05-30:
To clarify I noticed this on the contesting input fields.

phl0, 2022-05-30:
> To clarify I noticed this on the contesting input fields.

Fixed there with https://github.com/magicbug/Cloudlog/pull/1487/commits/2294e0caf67c30e8ee5287ab454088f662db495c. Also part of PR https://github.com/magicbug/Cloudlog/pull/1487 waiting for approval.

phl0, 2022-05-30:
PR https://github.com/magicbug/Cloudlog/pull/1487 has been merged. Keen to see if that solved your issues.

phl0, 2022-06-01:
@Cadair Please close here if my fix solved your problem 😇

closed 2022-06-13 by magicbug