magicbug/Cloudlog #1490: Installing fail when wizard setup windows 11

faridfarhan28, 2022-06-02:
Device : Legio 5 Pro
OS : Windows 11
WAMP : 3.2.4.2 64bit
Browser : Opera 87.0.4390.36

hi, i try to wizard install localhost/cloudlog/install but after input and click install, it stuck as screenshot.
thanks

[image]

phl0, 2022-06-02:
Younhave to create a database user that cloudlog can use to store its data to the MySQL db. You obviously missed that. This is required so that cloudlog can create its tables during install.

magicbug, 2022-06-13:
As @phl0 pointed out, this is an issue with MySQL username/password once you have this fixed should install fine