magicbug/Cloudlog #1500: Give summary of the number of grid squares worked

g4klx, 2022-06-19:
It would be nice if at the top of the grid square map, or somewhere else, that there could be a line summarising the number of squares claimed to have been worked, alongside the number confirmed.

phl0, 2022-06-20:
How about this?

[image]

[image]

https://github.com/phl0/Cloudlog/tree/addGridCountToMap

g4klx, 2022-06-20:
Spot on, exactly as I envisioned.

 
 

https://github.com/phl0/Cloudlog/tree/addGridCountToMap

magicbug, 2022-06-20:
Merged into the beta branch so it will filter across my hosted installs in the next hour or those who self host will need to git pull ;-)

phl0, 2022-06-20:
Updated my installation. Working fine here. Have fun with it @g4klx :-)
Think we can close here?

magicbug, 2022-06-20:
Think so, I'll close can always reopen at a later date