magicbug/Cloudlog #1502: eQSL QSO Upload is not uploading all QSOs

do4bmw, 2022-06-27:
Hy

I have a problem that the eQSL upload dont upload all my "old" QSOs

and when i click in Tools on "Mark all QSOs as Send"

these list from the OLD QSOs are still there

maybe someone can help me

mny thanks
[image]
[image]
[image]

magicbug, 2022-06-27:
Interesting, are they a part of the same active location that you're using?

do4bmw, 2022-06-27:
i think so ... 

[image]

do4bmw, 2022-06-27:
one more thing

when i click on these QSOs and EDIT them ...

in ALL of them the mode ist set to AM 

see screenshots, is there a way to modify them all at once ? or do i need to edit all of them (around 900 QSOs)

[image]
[image]
[image]
[image]

do4bmw, 2022-06-27:
and FUN FACT all these QSOs are Confirmed from eQSL and LOTW ! 
but still in my upload list ... is there a ERROR LOG for the upload that i can look at ?

magicbug, 2022-06-28:
If you go to QSL and check the eQSL tab in the edit qso are they marked as sent?

do4bmw, 2022-06-28:
that works ... but it will be a pain in the ass todo this for nearly  ~900 QSOs

do4bmw, 2022-06-28:
i dont need them all uploaded to eQSL ... i just want that list clear ... is there maybe a work around to mark them all as send , because the "Mark all as Send" Button is not working with these QSOs

do4bmw, 2022-07-04:
no help ? no infos :-( 

do i realy need to change all of these QSO by Hand ? :-(

AndreasK79, 2022-07-04:
@do4bmw are you using the master branch or the station_logbooks branch?

AndreasK79, 2022-07-04:
@do4bmw pull the latest code and try to mark all QSOs now.

AndreasK79, 2022-07-04:
@do4bmw now you can pull the latest code. I've refined it, so only those QSOs that are not marked with Y as sent will be updated to sent.

do4bmw, 2022-07-05:
doing update now ... and try to upload and after that , try to mark the QSO as send

will return in a few minutes

do4bmw, 2022-07-05:
YAY !!! the list is empty

Many Many Thanks !

[image]

closed 2022-07-06 by do4bmw