magicbug/Cloudlog #1503: Contest mode on two different machines

m0urs, 2022-06-27:
Last weekend I tried to log contest QSOs from both my desktop machine and my iPad. I started contest logging on my iPad and wanted later to continue logging on my desktop PC.

However, although all logged QSOs from the iPad appeared in the logbook correctly, I was unable to see the contest QSO in the contest logbook on my desktop. There was just an empty list. And the duplicate check was also not working.

I then tried again on the iPad and everything worked fine here.

So it seems that the contest logging information is not stored somewhere in the database but somewhere locally (a cookie?)

Is this true? If yes, is this something which could be changed? If not, did I do something wrong?

AndreasK79, 2022-06-27:
You are right. It is not stored in the database. It is stored in local storage in the browser. That is the reason you can't switch like that. 

No plans exists to change that, but who knows what will happen in the future.

m0urs, 2022-06-28:
It's a pity that there are no plans to fix that. I often user my portable station e.g. in the garden to do some contest QSOs and then will continue later in my shack again. During working outside it would be enough to have my iPad for logging and not needing to move my computer outside.

I really like the contest logging mode in Cloudlog. As I only do QRP work, I am not a contest hardliner and just the basic functionality I have in Cloudlog would be sufficient for me.

Maybe this could become a feature request for some future version? ;-)

AndreasK79, 2022-06-28:
No one has reported this before. That's the reason, so it can probably be added. I'll make a note and think about how it can be done.

electricEngineer, 2023-04-08:
+1 from my side. I would really apreciate this feature!

int2001, 2023-04-10:
+1 from me also. Would be great, tapped into that a few times

AndreasK79, 2023-04-10:
I have added pull request #2061 where I moved the contest sessions from the browser to the database. Testing and feedback welcome.

m0urs, 2023-04-10:
Thanks so much. Will be able to test it next week as I am currently on vacation!

int2001, 2023-04-11:
Great. Will test it as soon the PR ist merged into dev

m0urs, 2023-04-16:
@AndreasK79 I have just tested with my iPad and my Desktop PC and I can confirm that it works for me so far. Thanks ver ymuch for that enhancement!

closed 2023-04-16 by m0urs

AndreasK79, 2023-04-16:
Excellent :) Cabrillo export has been made as well. It's not merged yet, but will be in dev soon.