magicbug/Cloudlog #1504: Date on the Logbook page is incorrect

g4klx, 2022-06-28:
I'm using the hosted beta version of Cloudlog.

On the Logbook page, the Date column is correct for at least the first QSO on the page but that date is used for all of the QSOs on that page, and ones further back in time. If I go into "Edit QSO" the date is correct.

The image shows the date not changing even though the time shows that a day ended and a new one started.

This was produced just by going into the Log
[image]
book page.

phl0, 2022-06-28:
I see different dates on that picture?!

g4klx, 2022-06-28:
I feel like a fool! You are absolutely correct. I'll close the issue.

closed 2022-06-28 by g4klx

phl0, 2022-06-28:
No worries. I know the feeling. The answer usually drops from sky after I asked in a forum or the like :)