magicbug/Cloudlog #1508: [station_logbooks] Automatic Logging QSOs with Gridtracker is broken

m0urs, 2022-06-30:
It seems that with Version 2 Gridtracker can no longer log QSOs via the API. Although the "Test" button in Grditracker says "PASSED" no QSOs are logged,

It seems to use this URL:

`https://cloudlog.urspringer.de/index.php/api/qso`

Could it be that this is not an issue of Cloudlog but that Gridtracker needs to be adapted ? There is a similar issue with another application https://github.com/magicbug/Cloudlog/issues/1479

If this is something the other side needs to change: Is there a possibility to have kind of a "compatibility mode" in Cloudlog, so that old API calls are working a before (and you somewhere specify which user / logbook should be used for logging)?

I guess most of us are using Cloudlog with only one user and for them at least all third party tools would still work as before?

magicbug, 2022-06-30:
This isn't a Cloudlog issue, however I'm told it was sorted by passing a station location id.

closed 2022-06-30 by magicbug

m0urs, 2022-06-30:
Yes, that is what I thought. There is a 6month old issue with GridTracker (https://gitlab.com/gridtracker.org/gridtracker/-/issues/133) but no solution yet.

Gridtracker is not the only program which might to be adapted if Version 2 of Cloudlog gets productive and that is the reason why I asked if it would be possible to just define a setting in Cloudlog saying, if no station_id is passed via the API, then use this specified ID. 

It would get us at least the current functionality, otherwise all the programs are broken until they fixed that issue.

phl0, 2022-06-30:
> It would get us at least the current functionality, otherwise all the programs are broken until they fixed that issue.

Or never implementing the new APIs because it still works ;-)

m0urs, 2022-06-30:
Yeah, maybe. But it is better to have a basic functionality than no functionality at all ;-) I am using Gridtracker only because of the automatic Cloudlog feature. If I would have an alternative than I would use that.

I tried to use this [Python script](https://www.dg9vh.de/2021/01/25/digimodes-auf-dem-raspberry-pi-4/)  (which seems to give the station_id as parameter already)., but I end up with a 403 error and have no idea what is the reason for that. If I get that working I would no longer need Gridtracker.

m0urs, 2022-06-30:
This is what my script sends

`{'key': 'XXXXXXXXXXXXXXXXXXX', 'station_profile_id': '1', 'type': 'adif', 'string': '<call:6>DB0XXX<gridsquare:4>IM86<mode:3>FT8<rst_sent:3>-05<rst_rcvd:3>-26<qso_date:8>20220629<time_on:6>195515<qso_date_off:8>20220629<time_off:6>195626<band:3>17m<freq:9>18.101846<station_callsign:6>DG3NAB<my_gridsquare:6>JO40KD<tx_pwr:2>10<eor>'}
`
This is the URL it is sent to:

`https://cloudlog.urspringer.de/index.php/api/qso`

This is the error I get:

`Error: POST https://cloudlog.urspringer.de/index.php/api/qso/ Result: 403`

which is an "unauthenticated" error.

Do I need something more besides my API key and the "station_profile_id" with version 2 to authenticate?

phl0, 2022-06-30:
Generally not. The JSON seems to be correct at the first glimpse. But if we are talking about FT8 you might have a look at WSJT-X improved. The integrated a patch of me that adds native Cloudlog support to WSJT-X. See here: 
https://sourceforge.net/projects/wsjt-x-improved/files/WSJT-X_v2.6.0/Versions%20with%20Cloudlog%20support/

m0urs, 2022-07-01:
Danke. Es wäre schön, wenn von WSJT-X nicht so viele Forks rumliegen würden :-( Jeder Fork hat irgendwas, was man gerne hätte, kein Fork hat alles drin. Da ich schon WSJT-Z nutze, kann ich leider nicht auf WSJT-X_enhanced umsteigen :-(

m0urs, 2022-07-01:
Although I am not a NWJS developer , I tried to fix the issue with Gridtracker by myself and at least for me, it seems to work now :-) We will see if they accept my [merge request](https://gitlab.com/gridtracker.org/gridtracker/-/merge_requests/176) ...