magicbug/Cloudlog #1511: Unable to use hamqth callbook in latest station_logbooks branch

ofadam, 2022-06-30:
Despite having credentials set properly, I'm unable to use the hamqth callbook in the latest station_logbooks branch. QRZ.com still works as expected, though.

No error is displayed, but using the Live or Post QSO form results in nothing being looked up (not even older QSOs already logged in Cloudlog.)

AndreasK79, 2022-07-02:
The problem is related to the QRZ lookup. Edit your account and turn off this:
[image]

This need to be fixed so that it is ignored if HamQTH is chosen.

phl0, 2022-07-02:
Oopps. That is probably my fault. I will take a look.

phl0, 2022-07-03:
Found a way to fix this by implementing the picture lookup functions also for hamqth profiles. Tests and reports welcome.

closed 2022-07-07 by magicbug