magicbug/Cloudlog #1513: First letter of county name being dropped

ofadam, 2022-07-01:
In v2, the first letter of the county name is now being dropped on all states with the latest commit. 

Example: "SC,eaufort" instead of "SC,Beaufort"

ofadam, 2022-07-01:
I forgot to mention that this is with the QRZ callbook integration.

phl0, 2022-07-01:
Which branch are you using? And where exactly is the error occuring?

ofadam, 2022-07-01:
station_logbooks branch – you can see the problem when on the Logbooks page and popping up a QSO lightbox to see the county entry that was sourced from QRZ.com.

phl0, 2022-07-02:
Hmm not happening on my installation.

[image]

How can I reproduce this pls?

AndreasK79, 2022-07-02:
Can' reproduce this here either. Are you using the latest codebase from station_logbooks?

ofadam, 2022-07-02:
I am indeed. My US_counties.csv file is intact, yet this is what I see when accessing various screens in Cloudlog.

[image]

Strangely, in Live QSO, this is what I see for the county field after entering a callsign and having it pull from the callbook service – it's unavailable.

[image]

phl0, 2022-07-03:
I entered KE8PZN and get:

[image]

And if I lookup the QSO after that:

[image]

I have enabled the qrz.com API and it pulls the info from there. Is yours enabled or disabled?

ofadam, 2022-07-04:
With QRZ callbook enabled, went through the same exact steps as you – I currently only have a non-paid QRZ.com account. Could that be the difference?

magicbug, 2023-08-03:
As this is nearly a year old I'm closing

closed 2023-08-03 by magicbug