magicbug/Cloudlog #1515: 'Visit Public Slug:' link does not include '/visitor/'

PotatoesMakeChips, 2022-07-03:
file: /application/views/logbooks/edit.php
line: 64
[image]
[image]

Clicking the link results in the following error:
[image]

Adding /visitor/ resolves the issue:
[image]

I'll make a pull request when I can set up a dev environment if no one beats me to it. Unless there's a different way you want to solve the issue

phl0, 2022-07-03:
Tnx for finding this bug. In the sense of the rest of the code the fix should be like here: 545e7765a08c20b1afc6cb9f8c28be14d673ad5d.

magicbug, 2022-07-03:
Looks a perfect fix @phl0 PR away

phl0, 2022-07-03:
There you go: https://github.com/magicbug/Cloudlog/pull/1518

PotatoesMakeChips, 2022-07-03:
That fix works on my end

closed 2022-07-03 by magicbug