magicbug/Cloudlog #1517: [station_logbooks] Uploading to Clublog failed

m0urs, 2022-07-03:
With Version 2 I am no longer able to upload my QSOs to Clublog. I am using this URL

`curl --silent https://cloudlog.urspringer.de/index.php/clublog/upload/dg3nab`

There is no error message or other message while running this URL.

I assume something needs to be added to be compatible with the new station logbooks feature.

magicbug, 2022-07-03:
Station Logbooks shouldn't have broken Clublog as it wouldn't even be looking for it or at least shouldn't be so needs further debugging

m0urs, 2022-07-03:
I tried to do some debugging, however I do not have any knowledge about how this CodeIgniter stuff works ...

`$clublog_info['user_name' is correctly set to "dg3nab"`

However, it seems that "$station_profiles = $this->stations->all_with_count();" does not correctly return the station profiles. "$station_profiles->num_rows();" seems to be 0 and therefor the loop "foreach ($station_profiles->result() as $station_row)" is never executed.

I added the line 

` echo "station_profiles->num_rows() = ",$station_profiles->num_rows();`

right after

`$station_profiles = $this->stations->all_with_count();`

which gives me

`"station_profiles->num_rows() = 0"`

magicbug, 2022-07-03:
Perfect thanks for checking into it, I plan on ripping most of the Clublog code apart you might have spotted a new table for storing login info for third partys this will be used to assign a location to a station location and will allow it to work across multiple accounts rather than just the one with one PHP call.

It will also allow disabling of uploading if Clublog returns a login error so that their system doesn't block the IP address after too many calls :)

AndreasK79, 2022-09-19:
This is probably fixed in station_logbooks. I rewrote the cron job a few days ago.

Now it loops through the users with a username and password set. Then QSOs are uploaded.

Just call it like this cloudlog-url/index.php/clublog/upload

closed 2022-09-22 by magicbug

m0urs, 2022-09-22:
I can confirm, that it is working for me. Thank you very much!