magicbug/Cloudlog #1530: [V2] Public Station Logbook Slugs Broken w/ Multiple Logbooks

cjtheham, 2022-07-10:
**Describe the bug**
On a server with two station logbooks, when two public slugs are created, the second one directs to the original. Unclear if this is a bug with just multiple slugs, or if it is an issue because one slug is created by an Admin role and another is created by an Editor role.

**To Reproduce**
Steps to reproduce the behaviour:
1. Create two station logbooks on two users. Give each one a few contacts in their log.
2. Add any public slug to each logbook.
3. See what output is on the public url.

**Expected behaviour**
Expect to see the correct logbooks added to the station logbook and their data on the public slug, not another user's log.

**Screenshots**
Public Slug Data Shown
[image]

Correct Data for this Station Logbook
[image]

Running on Ubuntu 20.04 server, output appearing on latest versions of Chrome, Firefox, and Edge.

magicbug, 2022-07-10:
Yep somethings broke this will get it fixed :)

magicbug, 2022-07-10:
@cjtheham this should hopefully be resolved

commit 0fd28941 (2022-07-10): [Visitor] Hopefully fixes incorrect logs loading

Maybe Fixes #1530
closed 2022-07-18 by magicbug