magicbug/Cloudlog #1546: Problem importing an ADIF File

chemicalalien, 2022-07-26:
When I try to import an ADIF File, I always get the following error:

...................................................................
An uncaught Exception was encountered
Type: Error
Message: Call to undefined function mb_convert_encoding()
Filename: /var/www/html/application/libraries/Adif_parser.php
Line Number: 125
....................................................................

This error occurs regardless what kind of ADIF File I try to import.
Even reimporting a test qso that I exported from cloudlog results in the same error.

Line 125 of Adif_parser.php is a string conversion

magicbug, 2022-07-26:
Probably the mbstrings PHP extension is missing from the server setup

chemicalalien, 2022-07-26:
Yep, that solved the problem. mni tnx.

closed 2022-07-26 by chemicalalien