magicbug/Cloudlog #1548: eQSL confirmations don't show as grid confirmations

g4klx, 2022-07-27:
This is using the hosted version of the development version of Cloudlog.

If a QSO has been confirmed via LoTW then it appears to be completely confirmed, in particular the grid square count is updated and the grid square is shown in green.

However if the only QSO for a grid square has only been confirmed via eQSL then the grid square remains unconfirmed despite the arrows for eQSL both being green. The QSO is indeed confirmed with eQSL itself when I log into it.

Is this a bug or a feature?

magicbug, 2022-07-27:
Its by design, I should add a flag but as eqsl isn't accepted for vucc I just ignore it

g4klx, 2022-07-27:
It would be good to add it as a flag as it could/would still mean something to the operator though.

AndreasK79, 2023-06-29:
@g4klx this one is fixed in the dev branch. I rewrote the Gridsquare Map with more filters:
[image]

closed 2023-06-29 by AndreasK79