magicbug/Cloudlog #1549: Analytics could usefully use a page of statistics

g4klx, 2022-07-27:
It would be great to have a page in the analytics section that gave the raw statistics of activity across all of the bands. For example the list of bands in the left most column, followed by columns for the number of countries, grid squares, zones, etc. that have been confirmed (and maybe unconfirmed also). Only showing those bands where the numbers are greater than zero.

That way it would be easy to see progress towards awards, and general status of the confirmations. It wouldn't need any special graphics so would presumably be pretty efficient too.

kb-light, 2022-08-24:
I would be interested in number of unique stations (maybe also by band/mode).

kb-light, 2022-09-28:
I am fine with the new statistics from #1621

closed 2022-10-11 by magicbug