magicbug/Cloudlog #1561: Internet as "band" option (feature request)

SethPowsner, 2022-08-15:
I've had some QSOs over VBand / CW Hotline. Would be nice to have a way to indicate "internet" as the band (rather than some wavelength). Not sure if someone else will want DMR etc for phone. Maybe "internet" as the band would be sufficient for both. Not sure if there's anyway to export such an entry usefully. Maybe default should be not to export "internet" entries. Doubt it's worth adding an export control option at this time.

Thanks in advance,
Seth W1ETA

magicbug, 2022-08-15:
Currently we support having a band with a propagation of "internet" but wont be adding a "internet" band.

closed 2022-08-15 by magicbug

SethPowsner, 2022-08-15:
Hmm... Got the idea. Was able to find it now knowing to look for "propagation."
Thanks.