magicbug/Cloudlog #1566: Custom Map - Allow filtering by mode / propagation

magicbug, 2022-08-16:
### Discussed in https://github.com/magicbug/Cloudlog/discussions/1563

<div type='discussions-op-text'>

<sup>Originally posted by **LAwmand** August 16, 2022</sup>
2. On a custom map dates search could other search criteria be added such as mode to narrow down qso's such as "mode" or "propagation type"? So, for example I could just see my meteor scatter contacts in August rather than seeing everything.
</div>

AndreasK79, 2022-08-27:
Something like this:
[image]

AndreasK79, 2022-08-27:
#1583 created for testing.

AndreasK79, 2022-08-27:
And with some tweaking, I made the date selection the same size as the others:
[image]

AndreasK79, 2022-08-28:
It now keeps the datetimes too :)

phl0, 2022-08-28:
Cool. I will test and report.

phl0, 2022-09-02:
Forgot to report here. Works for me. Go for merge :)

closed 2022-09-14 by magicbug