magicbug/Cloudlog #1586: [API][QSO] QSO not included in ADIF export when LoTW state is NULL

kb-light, 2022-09-01:
**Describe the bug**
When using the Cloudlog API to upload QSOs from WSJT-X the LoTW received state is Null. These QSOs are not included in ADIF export, when using "Export QSOs not uploadad to LoTW".

Branch: station_logbooks

**To Reproduce**
Steps to reproduce the behaviour:
1. Setup WSJT-X improved PLUS and link to Cloudlog
2. Make a QSO
3. Verify that it got added to Cloudlog (LoTW state shows not uploaded)
4. Go to ADIF Import / Export -> ADIF Export
5. Select your Callsign and "Export QSOs not uploadad to LoTW"
6. The QSO is not included in the exported ADIF

You can also use Advanced Search and compare the results of following queries:
- LOTW_QSL_SENT equal Y
- LOTW_QSL_SENT equal N
- LOTW_QSL_SENT is null
- LOTW_QSL_SENT not equal Y (not including LOTW_QSL_SENT is null)

**Expected behaviour**
All QSOs that are not flagged as LoTW QSL sent are included within the ADIF export

**Desktop:**
 - OS: Windows 10
 - Browser: Firefox
 - Cloudlog: station_logbooks
 - WSJT-X: WSJT-X improved PLUS v2.6.0~rc3 (@phl0)

phl0, 2022-09-01:
Could be related to https://github.com/magicbug/Cloudlog/pull/1523. I will check.

phl0, 2022-09-01:
Ok, just confirmed that `COL_LOTW_QSL_SENT? in NULL upon uploading a QSO via API (from WSJT-X improved):

```mysql> select COL_QSO_DATE, COL_TIME_ON, COL_CALL, COL_LOTW_QSL_SENT from TABLE_HRD_CONTACTS_V01 where 1 order by COL_PRIMARY_KEY  desc limit 1;
+--------------+---------------------+----------+-------------------+
| COL_QSO_DATE | COL_TIME_ON         | COL_CALL | COL_LOTW_QSL_SENT |
+--------------+---------------------+----------+-------------------+
| 2022-09-01   | 2022-09-01 12:27:00 | OE3DXA   | NULL              |
+--------------+---------------------+----------+-------------------+
1 row in set (0.00 sec)
```

kb-light, 2022-09-01:
#1523 seems to fix the direct LoTW upload, but I am using the ADIF export. So `application/models/Adif_data.php` also needs the fix or the column should not be NULL.

phl0, 2022-09-01:
Yes yes. But I think it also makes sense to set the QSL columns to "N" instead on NULL for the API imported QSOs. What is @magicbug's opinion on this?

phl0, 2022-09-01:
@kb-light Maybe you can test this patch?

https://github.com/magicbug/Cloudlog/commit/44dd9a981e0190d708483e9cf4f9237ad3e7ebe8

It should also export QSOs with LOTW_QSL_SENT = NULL.

phl0, 2022-09-01:
I have created a migration that sets all QSL columns to N where the current value is NULL and then changes the column structure to be NOT NULL and 'N' as default value:

See: https://github.com/phl0/Cloudlog/commit/a937532cdcccaf9d646f361f4b81d233bbfbff6a

Need to test if this upgraded table structure would then save API QSOs also correctly with N instead of NULL for QSL columns.

phl0, 2022-09-02:
QSOs via API would not be saved because the QSL columns would be saved with NULL as value which the DB does not accept any more. So I refactored the code to set 'N' as default value if it is missing or NULL. The commit is here: https://github.com/phl0/Cloudlog/commit/6b31ffd9893290049a2a9df4c741301934e409fc.

Needs further testing if it does break QSO logging in other places.

phl0, 2022-09-02:
Created PR https://github.com/magicbug/Cloudlog/pull/1588 to fix this issue. Setting the columns to `NOT NULL DEFAULT 'N'` to be discussed separately.

phl0, 2022-09-07:
PR was merged. Should be fixed now.

closed 2022-09-14 by magicbug