magicbug/Cloudlog #1587: Google Bot / Bing Bot seems to cause a PHP error when indexing Cloudlog instances

dm5tt, 2022-09-02:
**Describe the bug**
Google Bot / Bing Bot seems to cause an PHP error when trying to index a Cloudlog Instance.

CloudLog sites are getting indexed with following error message

`A PHP Error was encountered. Severity: 8192. Message: Return type of CI_Session_files_driver::open($save_path, $name) should either be compatible with ...`

I'm not able to get the full mesage.

**To Reproduce**
1. Install CloudLog
2. Allow Google Bot / Bing Bot to index your Cloudlog Instace
3. Use Google/Bing and search for your Cloudlog instance

**Expected behaviour**
Not shoing any error message when getting hit by an internet crawler

**Screenshots**
 
[image]

magicbug, 2022-09-19:
Strange, can you tell us a bit about your install windows/linux/apache/php as I don't see this across all the ones I host but would be nice to fix it.

dm5tt, 2022-09-19:
Closing this as "Can't Repro".

closed 2022-09-19 by dm5tt