magicbug/Cloudlog #1606: Continuing problems with LOTW

ofadam, 2022-09-18:
On the latest station_logbooks branch, I've been getting error messages on every new QSO sent to LOTW since Sept. 14/15. It's actually happening on two separate installs with different callsigns and LOTW certs, and nothing has been changed to my knowledge.

Here's the error I get from LOTW:

2022-09-18 20:51:00 LOTW_QSO: Processing file: 20220918204557.924
2022-09-18 20:51:00 LOTW_QSO: User file: /var/www/kd9nye/html/uploads/lotw/kd9nye-2022-09-18-20-45-56-cloudlog.tq8
2022-09-18 20:51:00 LOTW_QSO: Certificate found for KD9NYE - UNITED STATES OF AMERICA (291)
2022-09-18 20:51:00 LOTW_QSO: Error in record 2: tCONTACT STATION_UID (1) doesn't match any tSTATION STATION_UID
2022-09-18 20:51:00 LOTW_QSO: : 2022-09-18 20:45:30Z  XXXXX 17M FT8
2022-09-18 20:51:00 LOTW_QSO: Record 2 ignored due to errors
2022-09-18 20:51:00 LOTW_QSO: Successfully processed 0 QSO records in 0.047063 seconds
2022-09-18 20:51:00 LOTW_QSO: 1 error encountered

ofadam, 2022-09-18:
I did find this similar issue, but the solution of re-uploading the LOTW cert didn't help.

https://github.com/magicbug/Cloudlog/discussions/802

magicbug, 2022-09-18:
Yep can confirm its currently not uploading to lotw

magicbug, 2022-09-18:
For me it broke 2022-09-16 10:02:55.. will need to investigate but will be in the morning not late at night :)

magicbug, 2022-09-18:
Not sure if this affects the master branch

magicbug, 2022-09-18:
Resolved it was PR https://github.com/magicbug/Cloudlog/pull/1603 that broke things by changing the EOR tags.

You'll need to mark QSOs sent to LoTW as not sent and reupload

closed 2022-09-18 by magicbug

ofadam, 2022-09-18:
Just tested and it is working yet again – thanks for the quick diagnosis!

phl0, 2022-09-19:
Err this awkward mistake was introduced by me. Sorry for that. I have to admit that I have to payed enough attention to the search and replace during patching: https://github.com/magicbug/Cloudlog/pull/1603#discussion_r973873612.
It is fixed and tested into a new PR: https://github.com/magicbug/Cloudlog/pull/1607.