magicbug/Cloudlog #1613: Cannot update DXCC data from Clublog

GregTroar, 2022-09-22:
I do not seem to be able to update country files.
Whenever I click on the Update DXCC Data button I have a "status:" appearing below and nothing else.

In the web server log I see that the file status.html does not exist in cloudlog/updates folder and cannot be reached.

Using Debian 11
Latest version of Cloudlog.

Error in the log:

*270 open() "/var/www/html/Cloudlog/updates/status.html" failed (2: No such file or directory)
*270 open() "/var/www/html/Cloudlog/update/dxcc" failed (2: No such file or directory)

Thanks

phl0, 2022-09-22:
The status.html should be there as it is part of the Cloudlog code and not generated dynamicalls. Has anything been deleted?

GregTroar, 2022-09-22:
Nope I just git clone and launch the install.
Nothing was deleted.

GregTroar, 2022-09-22:
In your current repo there is no status.html in updates folder btw

phl0, 2022-09-22:
There is:

[image]

Which branch are we talking about?

phl0, 2022-09-22:
Opps. Sorry, I mixed it up with index.html -.-

The status.html is created during update. So I guess the web server does not have write permissions to the directory. You may need to adjust permissions for www-data user to be able to write that file. There should be an error in the logs that it could not create the file prior to the failed access.

GregTroar, 2022-09-22:
That is not write issues as for testing purpose I did chmod 777 clouldlog directory.
Not sure what is wrong

phl0, 2022-09-22:
Hhmm, try setting the log threshold in `application/config/config.php` to 4 and have a look at the cloudlog logs.
Just tried to update here and it seems Clublog is not up to providing data:

[image]

AndreasK79, 2022-09-22:
It's working here now:
[image]

phl0, 2022-09-22:
QSL.

[image]

GregTroar, 2022-09-22:
in application/logs is empty

Here is what it does on my side

[image]

GregTroar, 2022-09-22:
I don't find any treshold in the config.php file can you point me to the right place ?

Anyway thanks a lot for your help !

AndreasK79, 2022-09-22:
application/config/config.php: $config['log_threshold'] = 0;

phl0, 2022-09-22:
Around line 305 in `application/config/config.php`

The parameter is:

```
$config['log_threshold'] = 0
```

Set this to 4.

GregTroar, 2022-09-22:
Maybe Nginx is the issue as I already had to change the index.php into index.php? in the config file ?!?

phl0, 2022-09-22:
> Maybe Nginx is the issue as I already had to change the index.php into index.php? in the config file ?!?

Don't think so. Should work with nginx.

GregTroar, 2022-09-22:
Ok checking, starting from scratch as well and I revert in 5 minutes

GregTroar, 2022-09-22:
Ok i started from scratch, had to change again the index.php to index.php?, I could create new user, I put the log treshold to 4, did a chmod 777 on cloudlog for testing, update country file and again nothing happens.

The log also shows nothing even after clicking on Update button

`INFO - 2022-09-22 07:30:16 --> Config Class Initialized
INFO - 2022-09-22 07:30:16 --> Hooks Class Initialized
DEBUG - 2022-09-22 07:30:16 --> UTF-8 Support Enabled
INFO - 2022-09-22 07:30:16 --> Utf8 Class Initialized
INFO - 2022-09-22 07:30:16 --> URI Class Initialized
INFO - 2022-09-22 07:30:16 --> Router Class Initialized
INFO - 2022-09-22 07:30:16 --> Output Class Initialized
INFO - 2022-09-22 07:30:16 --> Security Class Initialized
DEBUG - 2022-09-22 07:30:16 --> Global POST, GET and COOKIE data sanitized
INFO - 2022-09-22 07:30:16 --> Input Class Initialized
INFO - 2022-09-22 07:30:16 --> Language Class Initialized
INFO - 2022-09-22 07:30:16 --> Loader Class Initialized
DEBUG - 2022-09-22 07:30:16 --> Config file loaded: /var/www/html/cloudlog/application/config/cloudlog.php
DEBUG - 2022-09-22 07:30:16 --> Config file loaded: /var/www/html/cloudlog/application/config/bands.php
DEBUG - 2022-09-22 07:30:16 --> Config file loaded: /var/www/html/cloudlog/application/config/lotw.php
INFO - 2022-09-22 07:30:16 --> Helper loaded: url_helper
INFO - 2022-09-22 07:30:16 --> Helper loaded: security_helper
INFO - 2022-09-22 07:30:16 --> Language file loaded: language/english/general_words_lang.php
INFO - 2022-09-22 07:30:16 --> Database Driver Class Initialized
INFO - 2022-09-22 07:30:16 --> Session: Class initialized using 'files' driver.
DEBUG - 2022-09-22 07:30:16 --> cURL Class Initialized
INFO - 2022-09-22 07:30:16 --> Migrations Class Initialized
INFO - 2022-09-22 07:30:16 --> Language file loaded: language/english/migration_lang.php
INFO - 2022-09-22 07:30:16 --> Database Forge Class Initialized
INFO - 2022-09-22 07:30:16 --> Model Class Initialized
INFO - 2022-09-22 07:30:16 --> Controller Class Initialized
INFO - 2022-09-22 07:30:16 --> Language file loaded: language/english/qslcard_lang.php
INFO - 2022-09-22 07:30:16 --> Language file loaded: language/english/lotw_lang.php
INFO - 2022-09-22 07:30:16 --> Language file loaded: language/english/qso_lang.php
INFO - 2022-09-22 07:30:16 --> Model Class Initialized
INFO - 2022-09-22 07:30:16 --> Language file loaded: language/english/pagination_lang.php
INFO - 2022-09-22 07:30:16 --> Pagination Class Initialized
INFO - 2022-09-22 07:30:16 --> Model Class Initialized
INFO - 2022-09-22 07:30:16 --> File loaded: /var/www/html/cloudlog/application/views/interface_assets/header.php
INFO - 2022-09-22 07:30:16 --> File loaded: /var/www/html/cloudlog/application/views/view_log/partial/log_ajax.php
INFO - 2022-09-22 07:30:16 --> File loaded: /var/www/html/cloudlog/application/views/view_log/index.php
INFO - 2022-09-22 07:30:16 --> File loaded: /var/www/html/cloudlog/application/views/interface_assets/footer.php
INFO - 2022-09-22 07:30:16 --> Final output sent to browser
DEBUG - 2022-09-22 07:30:16 --> Total execution time: 0.0900
INFO - 2022-09-22 07:30:16 --> Config Class Initialized
INFO - 2022-09-22 07:30:16 --> Hooks Class Initialized
DEBUG - 2022-09-22 07:30:16 --> UTF-8 Support Enabled
INFO - 2022-09-22 07:30:16 --> Utf8 Class Initialized
INFO - 2022-09-22 07:30:16 --> URI Class Initialized
INFO - 2022-09-22 07:30:16 --> Router Class Initialized
INFO - 2022-09-22 07:30:16 --> Output Class Initialized
INFO - 2022-09-22 07:30:16 --> Security Class Initialized
DEBUG - 2022-09-22 07:30:16 --> Global POST, GET and COOKIE data sanitized
INFO - 2022-09-22 07:30:16 --> Input Class Initialized
INFO - 2022-09-22 07:30:16 --> Language Class Initialized
INFO - 2022-09-22 07:30:16 --> Loader Class Initialized
DEBUG - 2022-09-22 07:30:16 --> Config file loaded: /var/www/html/cloudlog/application/config/cloudlog.php
DEBUG - 2022-09-22 07:30:16 --> Config file loaded: /var/www/html/cloudlog/application/config/bands.php
DEBUG - 2022-09-22 07:30:16 --> Config file loaded: /var/www/html/cloudlog/application/config/lotw.php
INFO - 2022-09-22 07:30:16 --> Helper loaded: url_helper
INFO - 2022-09-22 07:30:16 --> Helper loaded: security_helper
INFO - 2022-09-22 07:30:16 --> Language file loaded: language/english/general_words_lang.php
INFO - 2022-09-22 07:30:16 --> Database Driver Class Initialized
INFO - 2022-09-22 07:30:16 --> Session: Class initialized using 'files' driver.
DEBUG - 2022-09-22 07:30:16 --> cURL Class Initialized
INFO - 2022-09-22 07:30:16 --> Migrations Class Initialized
INFO - 2022-09-22 07:30:16 --> Language file loaded: language/english/migration_lang.php
INFO - 2022-09-22 07:30:16 --> Database Forge Class Initialized
INFO - 2022-09-22 07:30:16 --> Model Class Initialized
INFO - 2022-09-22 07:30:16 --> Controller Class Initialized
INFO - 2022-09-22 07:30:16 --> Language file loaded: language/english/qslcard_lang.php
[... 11 lines cut ...]
INFO - 2022-09-22 07:30:20 --> Output Class Initialized
INFO - 2022-09-22 07:30:20 --> Security Class Initialized
DEBUG - 2022-09-22 07:30:20 --> Global POST, GET and COOKIE data sanitized
INFO - 2022-09-22 07:30:20 --> Input Class Initialized
INFO - 2022-09-22 07:30:20 --> Language Class Initialized
INFO - 2022-09-22 07:30:20 --> Loader Class Initialized
DEBUG - 2022-09-22 07:30:20 --> Config file loaded: /var/www/html/cloudlog/application/config/cloudlog.php
[... cut ...]
INFO - 2022-09-22 07:30:20 --> Controller Class Initialized
INFO - 2022-09-22 07:30:20 --> File loaded: /var/www/html/cloudlog/application/views/interface_assets/header.php
INFO - 2022-09-22 07:30:20 --> File loaded: /var/www/html/cloudlog/application/views/update/index.php
INFO - 2022-09-22 07:30:20 --> File loaded: /var/www/html/cloudlog/application/views/interface_assets/footer.php
INFO - 2022-09-22 07:30:20 --> Final output sent to browser
DEBUG - 2022-09-22 07:30:20 --> Total execution time: 0.0039`

phl0, 2022-09-22:
It takes a good while after clicking the button until further output appears on the page. Did you wait long enough? :-)
(the debug log is hard to read due to missing line breaks. Maybe you can re-format it).

phl0, 2022-09-22:
It seems that parts of the code are not executed. My debug log contains (after clicking the update button):

```
INFO - 2022-09-22 07:40:02 --> Model "DXCC_Entities" initialized
INFO - 2022-09-22 07:40:10 --> Model "DXCC_Exceptions" initialized
```

As far as I can see those lines do not appear in your log ....

GregTroar, 2022-09-22:
Yeah anyway I went through the log and do not see anything relevant.
I have waited up to one minute, but the point is this error in nginx log which bothers me:

`2022/09/22 14:40:53 [error] 732#732: *11 open() "/var/www/html/cloudlog/updates/status.html" failed (2: No such file or directory), client: 192.168.1.100`

I have hundred of these lines in the log

GregTroar, 2022-09-22:
I think i may need to try with apache2 to see if any difference

phl0, 2022-09-22:
I would like to reproduce it to find the error and fix it. Can you share some details of your setup please?

phl0, 2022-09-22:
> Yeah anyway I went through the log and do not see anything relevant. I have waited up to one minute, but the point is this error in nginx log which bothers me:
>
> `2022/09/22 14:40:53 [error] 732#732: [...]

Sure that is because the page tries to read the status information of the DXCC update. But as that is not getting written, it fails to read that file. Nothing else in nginx log before the no such file error appears?

GregTroar, 2022-09-22:
> I would like to reproduce it to find the error and fix it. Can you share some details of your setup please?

Let me know what info you need ?

I am on a virtual machine which is my web server running on Debian.
I have just done update and upgrade.

I am using php 7.4 but tried 7.0 and 8.0 (i know I should not but was for testing)

Everything I have in nginx log is:

[image]

GregTroar, 2022-09-22:
Ok I tried with Apache2 and remove the index.php? back to index.php and it works

The problem is that I would like to use Nginx as I have all my blogs/website running on it...

[image]

phl0, 2022-09-22:
> Let me know what info you need ?

Which database? mariadb? Which branch of Cloudlog? master or station_logbooks?

GregTroar, 2022-09-22:
Fyi also the CloudlogCAT will not sync with Nginx but works with Apache2.

Db: MariaDb yes and master branch

Let me know if you need further information I will stop testing for now and will come back from time to time to see if someone got it working with Nginx.

Thanks

phl0, 2022-09-22:
> Maybe Nginx is the issue as I already had to change the index.php into index.php? in the config file ?!?

Where did you add that to the nginx config exactly?

phl0, 2022-09-22:
Think I found it:

```
--- /etc/nginx/sites-enabled/default.orig	2022-09-22 10:06:32.616290461 +0100
+++ /etc/nginx/sites-enabled/default	2022-09-22 10:06:42.400240782 +0100
@@ -53,7 +53,7 @@ server {
 
 	# pass PHP scripts to FastCGI server
 	#
-	location ~ \.php$ {
+	location ~ \.php {
 		include snippets/fastcgi-php.conf;
 	
 		# With php-fpm (or other unix sockets):
```

By default nginx handles only PHP if files end on .php but not if the .php is part of the path like `http://domain.com/Cloudlog/index.php/update`

With the above change in my nginx config the DXCC update now works:

[image]

magicbug, 2022-09-22:
Think its safe to leave a wiki edit, this is down to a incorrectly configured Ngnix setup nice to see its working shall close the issue as its not a Cloudlog one

closed 2022-09-22 by magicbug

GregTroar, 2022-09-22:
It is indeed working now both the Cat sync and the update of DXCC thanks !

phl0, 2022-09-22:
Added a hint to the Wiki: https://github.com/magicbug/Cloudlog/wiki/Installation/448cdd443dc203b14f2d15adfa344135e29157d7