magicbug/Cloudlog #1614: [Version 2] Upgrade Checklist for users coming from V1

magicbug, 2022-09-22:
Thought I better write some notes down on what I was thinking for this as there are some functional changes.

- I'd suggest we use the config table to store whether the user has seen the V2 Checklist
- Need to suggest to user to take ownership of ALL Station Locations perhaps a button they can press
- API this needs similar treatment to Locations, or revoke ALL API keys and tell them to make new ones.
- Teach them about Logbooks and the importance and how to use them.
- Perhaps a little reminder to update the Country files
- In the config perhaps rename the user auth roles as you can now have standard users and admins with their own logbooks etc.

Theres probably some other bits, I'm doing Railways on the Air this weekend so times a bit thin but wanted to dump my brain!

magicbug, 2022-09-22:
### Changes to config.php

```php
$config['auth_level'][3] = "Operator";
$config['auth_level'][99] = "Administrator";
```

magicbug, 2022-09-22:
### Changes to index.php

Remind users that they might want to set the _ENVIRONMENT_ to production rather than development to hide errors and debugging data.

```php
define('ENVIRONMENT', 'production');
```

AndreasK79, 2022-09-25:
@phl0 pointed out that PHP 8.1 gives errors with Cloudlog. I think Codeigniter has been updated to fix this, but I need to test it out with an upgrade here. Is PHP 8.1 compatibility wanted for V2, or is 8.0 good enough for the time being?

magicbug, 2022-09-26:
8.1 would be nice but if we can scrape 8.0 that wouldn't be a bad thing :)

AndreasK79, 2022-09-26:
8.0 works. That was upgraded last year. I think the latest dev-version for Codeigniter is ok with 8.1, but need to test.

AndreasK79, 2022-09-26:
Just had a quick test with the latest dev version of CI. It does not fix the errors being thrown. That means 8.1 compatibility will have to wait.

magicbug, 2022-09-26:
> Just had a quick test with the latest dev version of CI. It does not fix the errors being thrown. That means 8.1 compatibility will have to wait.

Agreed

phl0, 2022-09-26:
> Just had a quick test with the latest dev version of CI. It does not fix the errors being thrown. That means 8.1 compatibility will have to wait.

Roger that.

magicbug, 2022-10-11:
OK @AndreasK79 @phl0 I believe this is now done, if you chaps can have a look

AndreasK79, 2022-10-11:
I'm stuck at welcome :)

AndreasK79, 2022-10-11:
Lovely colors too :)
[image]

Seems to be the code tag. Need to look into that as well.

AndreasK79, 2022-10-11:
Fixed a typo in the view, but it didn't fix my issue. Need to dig more later.

magicbug, 2022-10-11:
Probably my shockingly bad IF statements.. it will be a lesson of "don't code when feeling terrible" I'm sure of it :p

magicbug, 2022-10-11:
Poked a few more things and perhaps fixed, certainly isn't an issue with V1 to V2

ofadam, 2022-10-11:
Unfortunately, still seem to be stuck. Same issue with cron prompt not being able to be cleared in welcome message, but also with addition of API keys. After clicking on "Claim APIs" button and have it change to "All API Keys Claimed," clicking on "check if migration complete" button then protests with same API message.

AndreasK79, 2022-10-11:
That last commit solved it for me. I was having the exact same issue as @ofadam described above.

magicbug, 2022-10-11:
Perfect

ofadam, 2022-10-11:
Yep, I'm good now as well! Thanks for all your incredible work in getting to V2, all!

magicbug, 2022-10-11:
> Yep, I'm good now as well! Thanks for all your incredible work in getting to V2, all!

I'll be very happy once its live, can get back to fixing bugs and building new features ;-)

magicbug, 2022-10-11:
Right, sooo at some point I need to take the plunge and make this live once everyone thinks we are good to go

phl0, 2022-10-11:
> OK @AndreasK79 @phl0 I believe this is now done, if you chaps can have a look

I just set up another V1 instance, added some log data and then switch branches to `station_logbooks`. What shall I say? All migrations ran smoothly and the upgrade worked out of the box.

magicbug, 2022-10-11:
I'll let it sit for a night, see if anything appears, then do a quick run through everything to look for issues then perhaps its release time.

closed 2022-10-12 by magicbug