magicbug/Cloudlog #1618: US county is not re-enabled after field reset

phl0, 2022-09-24:
**Describe the bug**
The US county field is disabled upon entry of a non-US callsign. But it remains disabled after resetting the fields. So Logging a second QSO after that entry of US county or prefill of US county is not possible. Reloading the log entry page re-enables the field.

**To Reproduce**
Steps to reproduce the behaviour:
1. Enter a non-US callsign into the callsign field
2. Check the "General" tab
3. US County field is disabled (i.e. greyed out)
4. Press ESC or "Reset" button
5. US County field is not re-enabled

**Expected behaviour**
US County field should be re-enabled during field reset

**Screenshots**

[image]

phl0, 2022-09-24:
A patch for this behaviour can be found in https://github.com/magicbug/Cloudlog/pull/1619.

AndreasK79, 2022-09-24:
It is disabled because no state is filled. It does only check against that, not dxcc. And if you enable it, then press reset, it is still enabled. I remember that selectize.destroy would create trouble with some other js. I don't know what it was, but it was commented out in qso_edit in footer.php and also around line 2609.

AndreasK79, 2022-09-24:
It is also commented out in qso.js:58.

AndreasK79, 2022-09-24:
The only problem I see here, is that it is enabled after reset, so that it should not be enabled after reset. The reason why I did this, is that the state is needed to search correctly after county. There are so many counties, and many of them have the same name, just different state, so state if used to filter.

phl0, 2022-09-25:
I think it also has to do with https://github.com/magicbug/Cloudlog/issues/1615#issue-1382892278. The lookup function gives `callsign_usa_county` back. And there are two cases which obviously lead to different tesults. It can be an empty string "" which I think works correct. But there is also the case where it gives NULL and this leads to different behaviour (where it should not imho).

Need to look deeper into this why and where this differs. So my current patch my be inappropriate (though I think separate variable names to the selections would not harm).

closed 2023-05-08 by magicbug