magicbug/Cloudlog #1623: Feature Request: Add POTA functionality

zeidlos, 2022-09-28:
## Summary
The goal of this ticket is to add functionality needed to log POTA QSOs for both hunters and activators.
In here we can document and comment on what's needed to accomplish this goal.

## Why
As avid POTA activators, we'd really love to use Cloudlog to log our POTA activations and also allow hunters to log those QSOs.
Cloudlog already allows for logging on a mobile phone, so creating a specialized, simplified POTA logging option should not be a huge deal.

## TODOs

- [ ] Add neccesary fields to the database
- [ ] Adjust adif import/export
- [ ] Query POTA API for autocompletion of parks.
- [ ] Allow direct export/sync to POTA.app
- [ ] Simplified POTA logging interface, optimized for smartphones
- [ ] Possibly allow to self-spot after each QSO logged
- [ ] ...

phl0, 2022-09-28:
The current ADIF spec (v3.1.3) does not have separate fields for POTA (yet). However there has been a proposal for this. See https://groups.io/g/adifdev/message/8745. This can be added as soon as it is part of the ADIF spec imho.

magicbug, 2022-09-28:
> The current ADIF spec (v3.1.3) does not have separate fields for POTA (yet). However there has been a proposal for this. See https://groups.io/g/adifdev/message/8745. This can be added as soon as it is part of the ADIF spec imho.

I acting as Cloudlog supported POTA being added to the spec as well.. however of course once its in the standard these look like some solid suggestions.

magicbug, 2022-09-28:
> * [ ]  Possibly allow to self-spot after each QSO logged

I'm guessing this is some kind of spotting service that POTA has? some links to API docs would be super useful

phl0, 2022-10-05:
Prepared some stuff for this: https://github.com/phl0/Cloudlog/tree/POTA.
Mising:
- Update function for csv
- Bands support
- POTA Awards

phl0, 2022-10-05:
Updated:

    * [x]  Bands support
    * [x]  POTA Awards
    * [ ] Update function for csv

phl0, 2022-10-05:
According to the original list:

* [x]  Add neccesary fields to the database
* [x]  Adjust adif import/export
* [ ]  Query POTA API for autocompletion of parks.
* [ ]  Allow direct export/sync to POTA.app
* [ ]  Simplified POTA logging interface, optimized for smartphones
* [ ]  Possibly allow to self-spot after each QSO logged

* [x] Added support for bands
* [x] POTA Awards page

phl0, 2022-11-30:
Is there an option to pull a csv or similar with all valid park references?

phl0, 2022-11-30:
> Is there an option to pull a csv file or similar with all valid park references?

Found it: https://twitter.com/KI7QEK/status/1597969542675693569

So the updated list is:

* [x]  Add neccesary fields to the database
* [x]  Adjust adif import/export
* [x]  Query POTA API for autocompletion of parks.
* [ ]  Allow direct export/sync to POTA.app
* [ ]  Simplified POTA logging interface, optimized for smartphones
* [ ]  Possibly allow to self-spot after each QSO logged
* [x] Added support for bands
* [x] POTA Awards page

The autocompletion has been integrated just like SOTA and WWFF. There will be an update function to pull the list of parks.

Code for PR is here (now rebased to current dev branch): https://github.com/phl0/Cloudlog/tree/addPotaSupport

phl0, 2022-12-07:
With the new Release 2.2 as of today we now have basic support for the new POTA fields of ADIF 3.1.4.

magicbug, 2022-12-26:
Right to get things moving along with this some more what do we need todo for 

-  Allow direct export/sync to POTA.app
-  Simplified POTA logging interface, optimized for smartphones
-  Possibly allow to self-spot after each QSO logged (I'm guessing this is POTA Watch sort of thing)_

If hams who do POTA want to give us info we can move development :)

gnoshme, 2023-02-13:
So there are POTA fields in the 3.1.4 ADIF spec, but there's no mention of them in the POTA ADIF reference right now.  Not sure when there will be a switch, but right now POTA is using MY_SIG, MY_SIG_INFO, SIG and SIG_INFO for optional and required fields depending on the QSO type.

These are the two types of QSOs:
1. normal QSO with a hunter
2. Park to Park QSO with someone else in a park

The current Live QSO dialog works great for (1) and stores all the required fields for POTA.  For an activation without Park to Park contacts, an ADIF of Live QSO entries would work fine because the actual park reference isn't required - selecting it is part of the log upload process.

A park to park contact requires SIG: "POTA" and SIG_INFO: [Park reference] and that takes some clicking around and may also require a change of frequency.

I think an easy lift to support POTA and retain compatibility to QRZ/LoTW as most people also log POTA contacts to their normal logs is:
**POTA Station Log**
A POTA specific station log that for defaults SIG to "POTA" and exposes the SIG_INFO as the park reference lookup field. These fields can then be updated when POTA support ADIF 3.1.4

**POTA QSO interface**
The ability to set / change frequency / band / mode but the assumption that most of the time this is set and forget for an activation.
Callsign => RST(R) => RST(S).  Signal reports are pretty much the entire QSO dialog.

gnoshme, 2023-02-13:
So there are POTA fields in the 3.1.4 ADIF spec, but there's no mention of them in the POTA ADIF reference right now. Not sure when there will be a switch, but right now POTA is using MY_SIG, MY_SIG_INFO, SIG and SIG_INFO and MY_STATE for optional and required fields depending on the QSO type.

These are the two types of QSOs:

- normal QSO with a hunter
- Park to Park QSO with someone else in a park

I did a park yesterday and the current Live QSO dialog works great for (1) and stores all the required fields for POTA. For an activation without Park to Park contacts, an ADIF of Live QSO entries works fine because the actual park reference isn't required - selecting it is part of the log upload process.

A park to park contact requires SIG: "POTA" and SIG_INFO: [Park reference] and that takes some clicking around.  If someone hits you with a Park to Park then it's likely on the same frequency.  If you go hunting for other parks, then you'll have to change frequency / band / mode too.

I think an easy lift to support POTA while retaining compatibility to QRZ/LoTW as most people also log POTA contacts to their normal logs is:

**POTA Station Log**
A POTA specific station log that for defaults SIG to "POTA" and exposes the SIG_INFO as the park reference lookup field and assigns that to all logbook entries. When POTA supports ADIF 3.1.4 these can be switched to the POTA ADIF fields.

**POTA QSO interface**
The ability to set / change frequency / band / mode but the assumption that most of the time this is set and forget for an activation so secondary.  All other fields associated with a QSO are tertiary.
Callsign => RST(R) => RST(S). Signal reports are pretty much the entire QSO dialog.  That's the way I have my Live QSO setup right now.  Callsign (TAB populates from QRZ), enter report TAB, enter report ENTER. Next.
"Park to Park" field that is an alias to SIG_INFO field
ADIF should have SIG set to "POTA" (but not sure if ONLY when SIG_INFO is populated.

**Advanced Logbook additional filter(s)**
Something that makes it easy to isolate these QSOs in combination with the date fields, e.g. MY_SIG_INFO (park) or Station Location.  Anything to isolate for ease of ADIF creation and download.

gnoshme, 2023-02-13:
Yes. Saw some POTA references in the UI but but now understand it better, thanks for the link.  Like POTA REFERENCE in GENERAL tab is for park to park, right?  I missed that, but is POTA supporting ADIF 3.1.4 yet though?  If they are it's not in their upload spec.

closed 2023-02-21 by magicbug

gnoshme, 2023-02-23:
I confirmed with POTA that they do not yet support ADIF 3.1.4