magicbug/Cloudlog #1627: [Bug][station_logbooks] PHP Error when claiming ownership after migration

kb-light, 2022-10-01:
**Describe the bug**
When switching to branch station_logbooks and trying to claim ownership of a location a PHP error is displayed. The ownership is successfully claimed.

**To Reproduce**
Steps to reproduce the behaviour: 
1. Go to 'Station Location'
2. Click on 'Claim Ownership'
3. See error

**Expected behaviour**
No error is displayed.

**Screenshots**
[image]

**Server:**
 - Docker
 - Apache2
 - PHP 7.4
 - MariaDB 10

magicbug, 2022-10-01:
should be resolved now some left over code

closed 2022-10-01 by magicbug

kb-light, 2022-10-01:
Can confirm that it is resolved.