magicbug/Cloudlog #1630: "Undefined" error on Distances Worked page

ofadam, 2022-10-02:
On the latest station_logbooks branch, I get an "undefined" red error message when trying to plot the following bands on the Distances Worked page:

80m
40m
30m
20m
17m
15m
10m
6m

I've tried another installation with the same version of the station_logbooks branch and don't see any errors. Any idea what would make one work and the other not? I'm thinking it may be some bad distance data or gridsquares?

AndreasK79, 2022-10-03:
I've fixed several bugs in station_logbooks regarding this, but obviously issues exists. Without having access to the installation, it's a wild guess.

Do you have several station locations? Are the gridsquares set the same in both installations?

I'll take another look at the code, but at this point, it's difficult to know what to look for.

ofadam, 2022-10-03:
I do have several station locations (quite a few, due to Parks on the Air activations.) Those gridsquares are different, so maybe that's the issue? Now that I think about it, those are the bands I've used while mobile...

AndreasK79, 2022-10-04:
That's the thing. The code is validating both the station location and the qso gridsquare.

I tried to provoke an error in the qso and/or in the station location, and it was no problem here, so it's quite difficult to guess what the error might be.

magicbug, 2022-10-04:
I'll try see if i can find whats barfing on my end could be something obvious

AndreasK79, 2022-10-04:
Probably something stupid I'm not seeing here.

magicbug, 2022-10-04:
It seems to only do it on certain QSOs as I can narrow it down to a satellite qso subset! thus maybe something I can spot

magicbug, 2022-10-05:
This was an array offset error added some checks and now resolved

closed 2022-10-05 by magicbug