magicbug/Cloudlog #1631: Incorrect QSO totals for new statistics grid

ofadam, 2022-10-02:
In the latest version on the station_logbooks branch, both the X and Y axis QSO totals don't equal what's present in the table.

[image]

magicbug, 2022-10-02:
Believe this is a known issue with the dev branch but @AndreasK79 will be able to say for sure

AndreasK79, 2022-10-03:
Currently there are 2 issues with the code. It has to do with the data in the database.

1. Different casing on bands. 160m and 160M are treated as different bands.
2. COL_SUBMODE can be either null or blank. The code checks for null values.

If the database had been more strict, this wouldn't be an issue.

My bet is issue 2 is causing this.

It's on my list, just haven't gotten around to it.

AndreasK79, 2022-10-04:
@ofadam I've pushed a fix now. Could you pull the latest code and report back?

ofadam, 2022-10-04:
Works beautifully! Thanks!

closed 2022-10-04 by ofadam