magicbug/Cloudlog #1633: Reproduceable crash

LAwmand, 2022-10-04:
I know nothing about code.

The whole app crashes when I fully zoom out when displaying locator squares. I know it,s a stupid thing to do but ive twiddled my mousewheel the wrong way a few times. Once crashed it just freezes and is unresponsive until I hit the back button a couple of times
Windows 10 on Magicbug server, same error on my desktops and laptop

**To Reproduce**
Steps to reproduce the behaviour:
1. Go to "Analytics" then "Gridsquares"
2. Zoom out until it will zoom no more using mouse wheel
3. Once zoomed out nothing works, you can't zoom back or click anything or click headings
4. Back button gets you out of it after pressing it a few times

**Expected behaviour**
To zoom in and out flawlessly

**Desktop (please complete the following information):**
 - OS: Win 10
 - Microsoft Edge
 - Version

AndreasK79, 2022-10-04:
Hi. I've seen this issue myself. I think we need to specify max zoom levels to avoid this.

magicbug, 2022-10-04:
Quite likely, Leaflet gets a little upset if lots of markers are being displayed

phl0, 2022-10-04:
I think its not the markers but zoom level being 0 that causes the issue here. Up to zoom = 1 everything works fine on my end.

magicbug, 2022-10-04:
That narrows it down for a quick fix!

phl0, 2022-10-04:
This fixes it. Can someone test?

https://github.com/magicbug/Cloudlog/compare/station_logbooks...phl0:Cloudlog:minZoom?expand=1

phl0, 2022-10-04:
Fixed it by setting minZoom to 1. So you cannot zoom out more. See https://github.com/magicbug/Cloudlog/pull/1634.

phl0, 2022-10-04:
Custom map is not affected by the way. PR fixes it for gridsquares and activated gridsquares.

LAwmand, 2022-10-05:
Thanks for the prompt and effective response phl0.
Magicbug, will this be reflected on my site?
Regards all, David

closed 2022-10-05 by magicbug

LAwmand, 2022-10-05:
> should be reflected within the next hour just merged the fix into the
> codebase

Thanks Peter,
Best Regards
David