magicbug/Cloudlog #1640: [Feature][Bands] Disable columns function for awards

phl0, 2022-10-05:
During coding for POTA support I discovered that it might be useful to have a checkbox per column to disable all bands per award. E.g. all bands for POTA etc. @AndreasK79 What do you tink?

So like this roughly?

[image]

AndreasK79, 2022-10-05:
I like the idea. One "master" checkbox to disable all checkboxes for the award?

phl0, 2022-10-05:
> I like the idea. One "master" checkbox to disable all checkboxes for the award?

Exactly that.

AndreasK79, 2022-10-05:
Want me to look into it, or do you want to give it a go?

phl0, 2022-10-05:
Feel free to do it. I am just digging around trying to hide SAT pages and data if SAT is disabled in bands. See https://github.com/magicbug/Cloudlog/discussions/1637

AndreasK79, 2022-10-06:
Frontend is done and working:
[image]
Just need the backend to save the changes.

AndreasK79, 2022-10-06:
@phl0 here #1642 is your wish.

phl0, 2022-10-06:
> @phl0 here #1642 is your wish.

As you command my king. I will test as soon as possible an report.

closed 2022-10-11 by magicbug