magicbug/Cloudlog #1641: Don't mark QSOs as uploaded to LOTW if encounter error

ofadam, 2022-10-05:
**Describe the bug**
When faced with an invalid LOTW certificate, a user (me in this case!) may not know for a month or longer that their QSOs aren't being uploaded successfully.

**To Reproduce**
1. Hourly cron job fires to upload QSOs not marked as uploaded to LOTW
2. Next fire of job should pull any matches since that initial upload.

**Expected behaviour**
If Cloudlog receives an error from LOTW like this:

> 2022-10-05 02:16:23 LOTW_QSO: Processing ABORTED: INVALID CERTIFICATE - Certificate may not be used. Certificate status = Superceded

... don't mark any QSOs as uploaded to LOTW. It might be helpful to surface this messaging from LOTW somewhere on the /lotw/lotw_upload page that currently just shows QSO matches.

**Additional context**
Using the latest build of logbook_stations branch.

magicbug, 2022-10-11:
If i remember correctly LoTW when uploading doesn't return those errors, its either success or failed, its been a while since I checked so things might have changed, but its a bit more of a upload and cross fingers. will look into it to see if this has changed

magicbug, 2022-12-26:
Just to come back to this ARRLs server does not provide a method on "upload" to confirm if it accepted each qso only the file therefore this isn't possible

closed 2022-12-26 by magicbug