magicbug/Cloudlog #1644: ADIF Import PHP Error

ve1cwj, 2022-10-06:
**This is for reporting bugs with the code, if this issue is regarding setting up on your local server please use the forum see readme for details**

**Describe the bug**
While trying to import an ADIF file (previously created by TQSL) the upload fails with PHP Error as below...

```
Error: Adif_Parser Already Initialized or No in ADIF File
A PHP Error was encountered
Severity: Warning

Message: count(): Parameter must be an array or an object that implements Countable

Filename: libraries/Adif_parser.php

Line Number: 187
```

**Note: Running the "v2" station_logbooks branch

**To Reproduce**
Steps to reproduce the behaviour:
1. Go to 'ADIF Import/Export'
2. Select a Station Location
3. Click 'Choose File' button and select the .adi file
4. Click 'Upload' button
5. See error and no logs uploaded even though the page says "Yay, it's uploaded"

**Expected behaviour**
Expected the QSOs in the ADIF file to be loaded into the logbook with the selected Station Location

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: MacOS High Sierra 10.13.6
 - Browser Chrome 105.0.5195.125

**Additional Info**
Copy/past of the ADIF file below...

```
<CALL:5>F4DXV
   <BAND:4>70CM
   <MODE:2>FM
   <QSO_DATE:8>20221001
   <TIME_ON:6>112000
   <FREQ:7>435.250
   <BAND_RX:2>2M
   <FREQ_RX:7>145.960
   <PROP_MODE:3>SAT
   <SAT_NAME:5>AO-91
<EOR>
<CALL:5>EA1PA
   <BAND:4>70CM
   <MODE:2>FM
   <QSO_DATE:8>20221001
   <TIME_ON:6>112000
   <FREQ:7>435.250
   <BAND_RX:2>2M
   <FREQ_RX:7>145.960
   <PROP_MODE:3>SAT
   <SAT_NAME:5>AO-91
<EOR>
<CALL:5>EA5WA
   <BAND:4>70CM
   <MODE:2>FM
   <QSO_DATE:8>20221001
   <TIME_ON:6>112000
   <FREQ:7>435.250
   <BAND_RX:2>2M
   <FREQ_RX:7>145.960
   <PROP_MODE:3>SAT
   <SAT_NAME:5>AO-91
<EOR>
<CALL:5>DK9JC
   <BAND:4>70CM
   <MODE:2>FM
   <QSO_DATE:8>20221001
   <TIME_ON:6>112000
   <FREQ:7>435.250
   <BAND_RX:2>2M
   <FREQ_RX:7>145.960
   <PROP_MODE:3>SAT
   <SAT_NAME:5>AO-91
<EOR>
<CALL:6>2M0SQL
   <BAND:4>70CM
   <MODE:2>FM
   <QSO_DATE:8>20221001
   <TIME_ON:6>112000
   <FREQ:7>435.250
   <BAND_RX:2>2M
   <FREQ_RX:7>145.960
   <PROP_MODE:3>SAT
   <SAT_NAME:5>AO-91
<EOR>
<CALL:6>MM6OEC
   <BAND:4>70CM
   <MODE:2>FM
   <QSO_DATE:8>20221001
   <TIME_ON:6>112000
   <FREQ:7>435.250
   <BAND_RX:2>2M
   <FREQ_RX:7>145.960
   <PROP_MODE:3>SAT
   <SAT_NAME:5>AO-91
<EOR>
<CALL:5>G3VUD
   <BAND:4>70CM
   <MODE:2>FM
   <QSO_DATE:8>20221001
   <TIME_ON:6>112000
   <FREQ:7>435.250
   <BAND_RX:2>2M
   <FREQ_RX:7>145.960
   <PROP_MODE:3>SAT
   <SAT_NAME:5>AO-91
<EOR>
```

magicbug, 2022-10-06:
No headers in the file?

magicbug, 2022-10-06:
Or people could code their software properly..

phl0, 2022-10-06:
Are the headers mandatory? Off to read the specs ...

AndreasK79, 2022-10-06:
The answer is no, so Cloudlog is not following the standard:

IV.A.2. ADI File Structure
An ADI file begins with an optional Header followed by one or more Records:

http://www.adif.org.uk/313/ADIF_313.htm