magicbug/Cloudlog #1646: Feature request: API for ADIF exports

kconkas, 2022-10-06:
I would like to export my logs into ADIF files regularly, using a cron job, for backup purposes. At the moment this can be done only from UI. It would be nice if there were an API endpoint, secured using an API key, that could be used for this purpose.

m0urs, 2022-10-20:
@magicbug I thought this is something we do have aleady?

Backup Log to ADIF:

https://cloudlog.server.com/index.php/backup/adif

Backup Notes to TXT:

https://cloudlog.server.com/index.php/backup/notes

I am quite sure that this worked before via command line in a cron job by using e.g.

 curl --silent https://cloudlog.server.com/index.php/backup/adif &>/dev/null

however I just saw that my cron jobs seem no longer working!? :-( I cannot say since when this is not working but I am sure it worked some time ago.

Maybe it stopped after upgrading to 2.0?

pa8s, 2022-11-04:
@m0urs Since upgrading to 2.0 the curl command does not work here either as in it will only generate an empty logbook.adi. Maybe it is only possible by api since 2.0?

rayslinky, 2022-12-09:
Looks like this only works if you're authenticated. I can get a successful backup by grabbing a cookie from login, and then calling the adif backup:
`curl 'https://cloudlog.local/index.php/backup/adif' -X POST -H "Cookie: $(curl -s -D - 'https://cloudlog.local/index.php/user/login' -X POST -H 'Content-Type: application/x-www-form-urlencoded' --data-raw 'id=&user_name=LOGIN&user_password=PASSWORD'|grep Cookie|sed 's/.*: //;s/;.*//')"`

dg9vh, 2022-12-09:
Could this also be a solution for download and upload eQSL per cron?

Kim

magicbug, 2022-12-09:
> Could this also be a solution for download and upload eQSL per cron

Yes but wouldn't recommend as long term solution

AndreasK79, 2022-12-11:
@magicbug Cron job for backing up adif does not work because of line 7 and 8 in adif_data.php.

Should we remove these lines so that the cron job can run again, or do you have any other suggestion?

pa8s, 2022-12-22:
> Looks like this only works if you're authenticated. I can get a successful backup by grabbing a cookie from login, and then calling the adif backup: `curl [...]

If I execute the command above on my log I get a response containing: `The backup of your log completed successfully. The output can be found at: https://my-url/backup/logbook.adi` This gives me the impression the logbook is available for download, so far so good. When I directly follow previous command with `curl https://my-url/backup/logbook.api` I only get:
```
Cloudlog ADIF export
<ADIF_VER:5>3.1.4
<PROGRAMID:8>Cloudlog
<PROGRAMVERSION:3>2.1
<EOH>
```

My logbook is up to date with the dev branch at the moment of writing.

DJLax5, 2023-02-09:
Yeah I tried to run:
> Looks like this only works if you're authenticated. I can get a successful backup by grabbing a cookie from login, and then calling the adif backup: `curl [...]

But as @pa8s said, that does not work. I managed to get it to work in a shell script using the following commands.
First, get the login cookie (obviously adjust the domain, user and password): 
`cookie=$(curl -s -c - 'https://<domian>/index.php/user/login' -X POST -H 'Content-Type: application/x-www-form-urlencoded' --data-raw 'id=&user_name=USER&user_password=password' | grep ci_cloudlog | awk '{print $7}')`
Next, trigger the backup creation:
`adif_response=$(curl -s -b "ci_cloudlog=$cookie" https://<domain>/index.php/backup/adif) `
`notes_response=$(curl -s -b "ci_cloudlog=$cookie" https://<domain>/index.php/backup/notes)`

I then check weather the responses contain the word "successfully". Note, that checks the complete server response, maybe there is a better option, because if the source contains a comment "successfully" it will pass the check.
And lastly, you can download the files:
`curl -o backup/notes.xml -LJO https://<domain>/backup/notes.xml`
`curl -o backup/logbook.adi -LJO https://<domain>/backup/logbook.adi`

I don't think that the feature is missing, it may be a bit cumbersome but in my opinion it does what it should do.

AndreasK79, 2023-05-02:
I implemented backup with a read-only API key. It's currently in this PR #2103. You generate a read-only key, then you call it with #cloudlogurl#/backup/adif/#apikey#. It will then output all QSOs that belong to the user with the key.

I also gave the backup a unique name, so that it would not get overwritten.

m0urs, 2023-05-02:
Thank you very much for this fix. Works great for me.

 Just to confirm: It seems that it backs up all station locations, right? 
 
 So now we only would need a similar fix for backup of the Notes which is broken since 2.0 as well ;-)

AndreasK79, 2023-05-02:
Yes. It will backup all station locations thst belong to the user. It gets the user based on the API key provided.

Didn't know about notes. Can probably make same fix there.

closed 2023-05-03 by AndreasK79

m0urs, 2023-05-03:
Thanks a lot. Works for me!

pa8s, 2023-05-09:
Although closed, just an remark. Is it possible to supply a filename to the api? E.g. #cloudlogurl#/backup/adif/#apikey#/mylog.adi

When using the api functionality now in a bash script you have to figure out what timestamp is given to the backup file to be able to download it with wget or curl. (Or I'm missing something ;) ) Or if a filename is not possible, add an option for e.g. logbook-latest.adi

Thnx.

phl0, 2023-05-10:
> (Or I'm missing something ;) ) 

Yes :). I have written a demo here:

https://github.com/magicbug/Cloudlog/wiki/Backup

That should work to detect the filename and download the file.

> Or if a filename is not possible, add an option for e.g. logbook-latest.adi

This would again raise the risk of overwriting data. :) But the solution mentioned in the Wiki should do the job.

pa8s, 2023-05-10:
Indeed, I was missing something. The wiki is a bit static so I did not check it this time. Thanks for the batch example, exactly what I need.

devinberg, 2023-05-30:
I have 1 user, 2 station logbooks, and 6 station locations. I have linked one of the locations to my main logbook and the other five locations to the second logbook just to separate my home logbook from my other operating locations. @AndreasK79 indicated that all locations should be backed up with this method. However, for me, running the backup with the API key only downloads the active logbook. The other locations are left out.

Is there any way to backup all of my logs?

AndreasK79, 2023-05-30:
@devinberg the code is supposed to take all locations that belong to a user. I'll take another look at the code.

AndreasK79, 2023-05-30:
@devinberg sounds like you found the issue. I might have an error in the logic which fetches locations :)

AndreasK79, 2023-05-30:
@devinberg I pushed a fix into the dev branch now. Thanks for reporting and identifying the problem.