magicbug/Cloudlog #1647: LOTW error when syncing

ofadam, 2022-10-07:
When Cloudlog tries to upload new QSOs to LOTW, it returns errors like this: "2022-10-07 02:40:22 LOTW_QSO: Error in record 3: DIV STYLE="BORDER: ADIF field size overflow"

This is using the latest station_logbooks branch on two different installs.

phl0, 2022-10-07:
This looks like some HTML code is in the adif file?!

magicbug, 2022-10-07:
Yeah, makes me wonder whats in a field to cause that, certainly isn't a wide spread bug so this is a localised issue

ofadam, 2022-10-07:
Is there a way I can stop the actual uploading of the ADIF to LOTW so I can grab it from /uploads?

magicbug, 2022-10-07:
No, because its not uploading ADIF.

magicbug, 2022-10-07:
I'd think "2022-10-07 02:40:22" narrows it down to what QSO is the issue however

ofadam, 2022-10-07:
Sorry, should have said TQ8 instead of ADIF...

On one of the two installs having this issue, I've only had one QSO since the error began. I looked at the ADIF export for that QSO, the SQL database entry and the web front end – and everything looks completely normal.

Any other suggestions for troubleshooting? One thing I did do around the same time was upgrade from PHP 7.4 to 8.0. I wonder if I missed something?

magicbug, 2022-10-07:
upgrading won't make any difference

ofadam, 2022-10-09:
A quick update for all – it started happening with I upgraded to PHP 8.0. I reverted to PHP 7.4 and LOTW syncing is working again.

pa8s, 2022-12-25:
Ok, closed issue but I just checked my Cloudlog-generated tq8 file and it also contains HTML code that causes the log not to be accepted by LoTW. 

Using php81.

The HTML code are located in the fields below:

```
---snip---
<Rec_Type:8>tCONTACT
<STATION_UID:1>1
<CALL:5>xxxxx
<BAND:3>12M
<MODE:3><div style="border:1px solid #990000;padding-left:20px;margin:0 0 10px 0;">

<h4>A PHP Error was encountered</h4>

<p>Severity: 8192</p>
<p>Message:  strtoupper(): Passing null to parameter #1 ($string) of type string is deprecated</p>
<p>Filename: adif_views/adif_export.php</p>
<p>Line Number: 45</p>

</div>FT8
<FREQ:9>24.916659
<FREQ_RX:1>0

<QSO_DATE:10>2022-12-24
<QSO_TIME:9>16:00:30Z
<div style="border:1px solid #990000;padding-left:20px;margin:0 0 10px 0;">

<h4>A PHP Error was encountered</h4>

<p>Severity: 8192</p>
<p>Message:  Function openssl_free_key() is deprecated</p>
<p>Filename: controllers/Lotw.php</p>
<p>Line Number: 914</p>

</div><SIGN_LOTW_V2.0:xx:6>fdc70bcd2985b0e4520JustSomeRandomHex3c8fc73592dca38e41ba03ffdf448
---snip---
```

phl0, 2023-01-02:
I have made a fix for this in https://github.com/magicbug/Cloudlog/pull/1895. @pa8s could you cross-check this?

closed 2023-01-03 by magicbug