magicbug/Cloudlog #1649: [Bug][station_logbooks] WAS Map does not work properly

kb-light, 2022-10-07:
**Describe the bug**
When a specific band is selected on the WAS Award page, then "Show WAS Map" does not work (selection works, but the map is not shown). Mode selection seems to work most of the time.

**To Reproduce**
Steps to reproduce the behaviour:
1. Go to 'Awards -> WAS'
2. Click on 'Show WAS Map'
3. Map is shown correctly
4. Select a Band
5. Click on 'Show WAS Map'
6. No Map is shown
7. Select 'Every Band'
8. Click on 'Show WAS Map'
9. When repeating step 8 sometimes the map is not shown

**Expected behaviour**
The Map should be shown.

**Screenshots**
Everything fine:
[image]

No map:
[image]

**Desktop:**
 - OS: Windows 10
 - Browser: Firefox 105.0

**Notebook:**
 - OS: Debian
 - Browser: Firefox 102.3
   - Less issues than on Windows
 - Browser: Chromium 105.0
   - More issues than on Firefox, but still better than the Windows PC

magicbug, 2022-10-07:
Is the developer tools console kicking out any errors? Firefox regardless of platform should in theory be exactly the same, as for Chromium can't recreate these errors at all.

magicbug, 2022-10-07:
Just spent some time on this and can't recreate it on Windows using Firefox, Edge, Chrome so i'd suggest running developer tools F12 in the browser and check the console tab and look for any errors.

rayslinky, 2022-10-08:
I have the same issue, and usually the map is not shown. I can occasionally refresh and get it to appear.
[image]

User agent:"Mozilla/5.0 (X11; Ubuntu; Linux x86_64; rv:105.0) Gecko/20100101 Firefox/105.0"

Warnings include:
```
This page uses the non standard property “zoom”. Consider using calc() in the relevant property values, or using “transform” along with “transform-origin: 0 0”. [was](https://home.rayslinky.com/qrz/index.php/awards/was)

InstallTrigger is deprecated and will be removed in the future. [utils.js:30:22](moz-extension://a8d22ed4-ce88-452d-b522-983f6754cac5/lib/utils.js)

Some cookies are misusing the recommended “SameSite“ attribute 2
Cookie “ci_cloudlog” does not have a proper “SameSite” attribute value. Soon, cookies without the “SameSite” attribute or with an invalid value will be treated as “Lax”. This means that the cookie will no longer be sent in third-party contexts. If your application depends on this cookie being available in such contexts, please add the “SameSite=None“ attribute to it. To know more about the “SameSite“ attribute, read https://developer.mozilla.org/docs/Web/HTTP/Headers/Set-Cookie/SameSite [All](https://home.rayslinky.com/qrz/index.php/awards/was_map/All/All)
Cookie “ci_cloudlog” does not have a proper “SameSite” attribute value. Soon, cookies without the “SameSite” attribute or with an invalid value will be treated as “Lax”. This means that the cookie will no longer be sent in third-party contexts. If your application depends on this cookie being available in such contexts, please add the “SameSite=None“ attribute to it. To know more about the “SameSite“ attribute, read https://developer.mozilla.org/docs/Web/HTTP/Headers/Set-Cookie/SameSite
```

rayslinky, 2022-10-08:
Oh, interestingly chrome gives me an error:
```
DevTools failed to load source map: Could not load content for https://site_fqdn/assets/css/bootstrap-dialog.css.map: HTTP error: status code 404, net::ERR_HTTP_RESPONSE_CODE_FAILURE
```

magicbug, 2022-10-08:
.map not to worried about thats CSS related

If anyone fancies giving me a copy of their database for testing purposes I might be able to get to the bottom of it :)

rayslinky, 2022-10-08:
> .map not to worried about thats CSS related
> 
> If anyone fancies giving me a copy of their database for testing purposes I might be able to get to the bottom of it :)

Sent to your qrz email. :)

magicbug, 2022-10-08:
Thanks @rayslinky that will make it easier to debug :)

magicbug, 2022-10-09:
Okay just had time to check with @rayslinky's database and I cannot recreate this within the [station_logbooks] branch aka the next version, so I'm thinking is perhaps isn't a code issue but a PHP timeout issue in the background.

I'd suggest perhaps enabling Codeigniter logging features in application/config/config.php and seeing if at some point it kicks out something useful relating to the problem, but certainly cant see any issues in the code which is promising at least

closed 2022-10-11 by magicbug