magicbug/Cloudlog #1657: Non-admins can edit bands for all users

Hubok, 2022-10-12:
I'm currently running eecd2bf71d944d24199c28162a3f535e051a8d3b, and while operators cannot see or edit custom bands other users add, they CAN edit the default bands and this reflects for all users.

I was able to edit the 160m band to be MF on an operator user, and another user sees the change.

As a sidenote, any chance to be able to reorder the bands or add bands for all users to use?

magicbug, 2022-10-12:
@Hubok Should be sorted now that only the admin sees the edit options

As for reorder this currently isn't an option

closed 2022-10-12 by magicbug

AndreasK79, 2022-10-14:
Delete should be available. It only deletes in bandxuser for the user, not for everyone. The band is not removed from the master band-table.