magicbug/Cloudlog #1659: Feature Request: Common Log for different OPs

myzinsky, 2022-10-12:
Im using CL with the station_logbook branch. And I thought that it's now also possible that logs can be shared among differnt OPs. But when I tried to test this, i saw that this is not possible. 

Following situation happened recently: 

We activated two special event calls with two operators on shortwave and QO100. We switched from time to time the callsigns between shortwave and QO100. For this purpose it would be awesome to have to personal logins for the operators and two shared logbooks were both could log into.

magicbug, 2022-10-12:
Code behind the scenes will support this, but currently having a logbook shared between club members isn't an option but on the roadmap

Hubok, 2022-10-13:
How would we do that before the feature gets implemented? I'm guessing create a station location per user, add them all to a station logbook, then manually edit the owner of the location with SQL queries?

myzinsky, 2022-10-13:
that could work yes

electricEngineer, 2024-02-05:
I'm trying to get logbook location sharing running. As I've seen, this way seems to be the easiest one with the current codebase and sql structure, specifically the table station_logbooks_relationship. @magicbug do you have already any plans or hints regarding this feature?

electricEngineer, 2024-02-07:
For all that are interested in shared logs I implemented most of the code that is needed to share a location because we really needed the feature at our club station. Mostly everything is working, but the sharing process has to be done still through SQL queries. But if anyone wants to give it a try you can find it there:
https://github.com/electricEngineer/Cloudlog_DL0TUH/tree/shared_logbook_locations

If anyone tried please let me know if there are any bugs.

The SQL query looks like this:
``` SQL
INSERT INTO station_logbooks_relationship (station_logbook_id, station_location_id) VALUES <sharing recipients logbook ID> <station location that should be shared>
```
I'm planning to also implement the sharing process in the location settings but that has to wait a bit. When all is done I will make a pull request.

gx1400, 2024-03-27:
I'm proposing using Cloudlog for our club logbook and this exact use-case came up to avoid creating a user and sharing a password. +1 in support of this feature.

patrickrb, 2024-03-27:
[image]

closed 2025-11-22 by magicbug