magicbug/Cloudlog #1662: Admin / Modes / FM shows SSB

antonov124, 2022-10-12:
Hi team,

Admin /Modes / FM shows "SSB" - Voice or similear

The selection menu offers SSB / DATA / CW. 

[fm_ssb](https://user-images.githubusercontent.com/27861984/195406363-85cceca8-5378-452c-b674-296949c37910.png)

"Error" in:  Cloudlog V1.xx and V2.xx also

phl0, 2022-10-12:
I guess we should categorize as VOICE / DATA / CW instead of SSB /DATA / CW. "FM" in category "SSB" causes irritation.

dg9vh, 2022-10-12:
What is the deeper sense of this classification SSB (new: Voice)/CW/DATA? We do not use it in the code on any place, right? I did not find it where it would be relevant. It was implemented in 2020 from the existing code as we went to make it customizable via database table.

But where is this classification used?

magicbug, 2022-10-12:
I can make this easy, won't be changed

closed 2022-10-12 by magicbug

dg9vh, 2022-10-12:
But what is it used for or planned to be used for?