magicbug/Cloudlog #1668: new update killed public access

kd4six, 2022-10-14:
**This is for reporting bugs with the code, if this issue is regarding setting up on your local server please use the forum see readme for details**

**Describe the bug**
going to the page now loads a log-in screen, instead of the public access version of the log

**To Reproduce**
Steps to reproduce the behaviour:
1. Go to my logbook: https://kd4six.cloudlog.co.uk

**Expected behaviour**
load the publicly accessible log / search page, without having to log in.

magicbug, 2022-10-14:
Not a bug, correct behaviour.

- Create a Station Logbook
- Assign it a public slug 
- on the main station list you'll then see a globe that is now your public URL for the logbook
- 
[image]

closed 2022-10-14 by magicbug

kd4six, 2022-10-14:
OK, I just did this, and it does show a publicly accessible page, but no search feature on that page? I'm doing something wrong, I'm sure :(