magicbug/Cloudlog #1671: Band Textboxes are showing a lot of bands

ea5wa, 2022-10-14:
When I open whatever band textbox, it shows a lot of bands....
[image]

magicbug, 2022-10-14:
I'd suggest deleting the duplicate bands seems they are listed twice.

ea5wa, 2022-10-14:
Yes, it works... But why has it happened?

AndreasK79, 2022-10-14:
Can't say I understand why. Migration checks if tables exists or not.

magicbug, 2022-10-14:
I'm going to close this as an unknown random bug and hope it doesn't reappear

closed 2022-10-14 by magicbug

AndreasK79, 2022-10-14:
We can hope. Impossible to know what happened without having the database before migration.

ea5wa, 2022-10-15:
Ok. Thanks

tonyc, 2023-05-02:
Just wanted to note, this is currently happening in my cloudlog-hosted instance.