magicbug/Cloudlog #1680: Remove slug from logbook not possible

PD0RZH, 2022-10-15:
**This is for reporting bugs with the code, if this issue is regarding setting up on your local server please use the forum see readme for details**

**Describe the bug**
Unable to remove a slug completely.

**To Reproduce**
Steps to reproduce the behaviour:
1. Go to Station Logbooks
2. Edit a logbook
3. Set a public slug and Save
4. Try to remove the slug (textfield empty)
5. System responds with "Looks good! is available"
6. Try to Save
7. Nothing happens, no error is shown

**Expected behaviour**
I expect the system can handle a default  (read empty) slug. 

**Desktop (please complete the following information):**
 - MAC OS Monterey
 - Safari
 - Version 16.0

dg9vh, 2022-10-15:
Use this link:
[image]

PD0RZH, 2022-10-15:
Isn't that the place to remove the logbook? I just want the slug to be removed.

magicbug, 2022-10-15:
Thats a bug.. I'll add an option later :)

phl0, 2022-10-15:
I have some basically working code for this:

[image]

Needs some CSS align on the buttons and some magic to make the input field empty as soon as the slug is removed.

magicbug, 2022-10-15:
If you want me todo the CSS magic put it as a PR and I'll do my thing later on it ?)

phl0, 2022-10-15:
Just had to take a break. Let me look into it once again. If I fail I will give a shout.

phl0, 2022-10-15:
I made a proposal: https://github.com/magicbug/Cloudlog/pull/1688

[image]

closed 2022-10-15 by magicbug