magicbug/Cloudlog #1681: Map not visible on Public View (using slug) and no pagination

PD0RZH, 2022-10-15:
**This is for reporting bugs with the code, if this issue is regarding setting up on your local server please use the forum see readme for details**

**Describe the bug**
Map not visible on Public View (using slug) and no pagination

**To Reproduce**
Steps to reproduce the behaviour:
1. Go to public view
2. No map and only 18 QSO's visible although on the right hand side you'll see the actual Total exceeding that number. No pagination on the bottom of the page

**Expected behaviour**
1. A visible map
2. Pagination

**Desktop (please complete the following information):**
 - MAC OS Monterey
 - Safari
 - 16

magicbug, 2022-10-15:
This was a bug, but was resolved yesterday afternoon in the code base

closed 2022-10-15 by magicbug

PD0RZH, 2022-10-15:
Peter, Should there be pagination as well on there Public view?