magicbug/Cloudlog #1686: On editing QSO it's not possible to add WWFF data

PD0RZH, 2022-10-15:
**This is for reporting bugs with the code, if this issue is regarding setting up on your local server please use the forum see readme for details**

**Describe the bug**
On editing QSO it's not possible to add WWFF data

**To Reproduce**
Steps to reproduce the behaviour:
1. Go to a QSO
2. Click on Edit QSO
3. Click on Awards
4. Paste GFF-0107 in the WWFF textbox
5. Click Save
6. Nothing happens

**Expected behaviour**
I expect it to save the WWFF reference in the WWFF field. Or perhaps I'm mistaking about the use of the WWFF field. Because in other cases I can see the SIG and SIG info field being used.
When using QSO > Post QSO the field populates accordingly expectations

**Screenshots**
[image]

[image]

**Desktop (please complete the following information):**
 - MAC OS Monterey
 - Safari
 - 16

magicbug, 2022-10-15:
Yep can confirm this is a bug you cant change the WWFF field when editing.

.. waves at @phl0

phl0, 2022-10-15:
Roger. Please assign to me (cannot do that myself). Will have a look. Tnx for reporting this.

phl0, 2022-10-15:
Tnx @AndreasK79. Seems this is not a bug but the operator missed to install the necessary WWFF data as per update function. See: https://github.com/magicbug/Cloudlog/wiki/Recommended-Cron-Jobs. 

I can reproduce the issue on my end if the WWFF data has not been loaded. But as soon as the data is loaded I can edit the QSO as requested:

[image]

Please update WWFF data and try again. If solved please close here. Thank you.

magicbug, 2022-10-15:
Confirmed this resolves it

Run the background task which is `/index.php/update/update_wwff` and it resolves it.

closed 2022-10-15 by magicbug