magicbug/Cloudlog #1687: Gridline qsos not properly showing up in gridsquare map for 6 meters

N9EAT, 2022-10-15:
I have multiple qsos that do not properly populate on the grid square map on 6 meters that are grid line qsos. This is an issue I do not have for satellite qsos. Grids are properly inputted into the vucc gridsquare field. The grids are still counted in the vucc total, and where the square is is still clickable and shows the qso in question. 

[image]

magicbug, 2022-10-15:
Thanks for reporting @N9EAT this is a bug, seems vucc squares for band isn't checked hence ignored.

AndreasK79, 2022-10-22:
#1714 fixed this problem.

AndreasK79, 2022-10-27:
This has been fixed in dev.

closed 2022-10-27 by AndreasK79