magicbug/Cloudlog #1702: WWFF: Autofill locator and QTH field

magicbug, 2022-10-18:
### Discussed in https://github.com/magicbug/Cloudlog/discussions/1701

<div type='discussions-op-text'>

<sup>Originally posted by **m0urs** October 18, 2022</sup>
I really like the separate WWFF field which has been introduced some time ago. It would be even better, if, after entering a WWFF reference, the locator and QTH field would be automatically filled, if not already populated. Currently I do these steps every time manually and it would be great help, if this could be done automatically.

Just like it has  already been implemented with the SOTA reference.

All the required information would be available as a CSV file from https://www.cqgma.org/wwff.csv

I already looked a bit into the code but that is, unfortunately, too difficult for me to do by myself :-(

</div>

phl0, 2022-10-19:
Assign it to me (I can't).

phl0, 2022-10-19:
For SOTA it is done by querying the SOTA API for summit details not from the SOTA txt file. I am checking if such exists for WWFF.

phl0, 2022-10-19:
wwff.co does not seem to have such a function. But I was able to provide some lines of code to cqgma.org which can now deliver such data. Cloudlog changes in https://github.com/magicbug/Cloudlog/pull/1704.

m0urs, 2022-10-20:
@phl0 I have two things which could be improved (from my point of view :-)):

1. If there is already a value in Locator or QTH, the auto fill option should not overwrite it.
2. The auto lookup (for SOTA and WWFF) seems only to work for a new QSO but not if I edit a existing QSO. Could we have that for edited QSOs as well?

phl0, 2022-10-20:
> 1. If there is already a value in Locator or QTH, the auto fill option should not overwrite it.

We now have 3 items that overwrite the grid locator besides it being manually entered or pulled from qrz/hamqth. That makes it 6 in sum. Which one should overwrite which one? 
I think it should stay the way it is: After you enter a callsign the grid is pulled from qrz (which most likely is the home grid). If you then enter a WWFF/SOTA reference it will be replaced. I think SOTA/WWFF grid should have preceedence as most ops most probably do not have their /p grids in qrz/hamqth. And if they give a grid during the QSO it should most likely be the same as the lookup function provides.

> 2. The auto lookup (for SOTA and WWFF) seems only to work for a new QSO but not if I edit a existing QSO. Could we have that for edited QSOs as well?

I think we could implement that.

phl0, 2022-10-26:
PR https://github.com/magicbug/Cloudlog/pull/1704 has been merged to `dev` branch. Pls test.

closed 2022-10-26 by magicbug

m0urs, 2022-10-26:
> PR #1704 has been merged to `dev` branch. Pls test.

This works for me. Thanks!